donalffons/opencascade.js #2: Compatibility issues

Johnly1986, 2020-07-23:
https://github.com/donalffons/opencascade.js/blob/4f185274832e1b814d07f51b6683aa68c4f5c926/make.py#L180
This option conflicts with “TOTAL_MEMORY“
https://github.com/donalffons/opencascade.js/blob/4f185274832e1b814d07f51b6683aa68c4f5c926/make.py#L183
https://github.com/donalffons/opencascade.js/blob/4f185274832e1b814d07f51b6683aa68c4f5c926/make.py#L184
These options are obsolete in new version

donalffons, 2020-07-23:
You're right, there are some issues with newer Emscripten versions. The last version that I tested this with is `1.39.11`. Therefore you might want to use this version for the moment. Under linux you configure Emscripten to use that version using
```
./emsdk install 1.39.11
./emsdk activate 1.39.11
source emsdk_env.sh
```
The windows commands should be similar.

Johnly1986, 2020-07-23:
Error: No tool or SDK found by name '1.39.11'.

Johnly1986, 2020-07-27:
I tested it in the docker version and an error occurred：
shared:ERROR: '/emscripten/upstream/bin/wasm2js --emscripten -O ./js/opencascade.wasm --mvp-features' failed (-9)
https://github.com/donalffons/opencascade.js/blob/2af3f532e14126e9ab35257ef07eb1fe28bd8884/make.py#L276

donalffons, 2020-07-28:
OK, so I finally got a hold of a windows laptop and tried building the library from there. And I could reproduce that error. This is pretty weird, because the container should normally not behave differently when you're on a different host system.

But anyway, here are two options that you can use to create your own builds with windows as your host system:
1. You can just comment [Line 35](https://github.com/donalffons/opencascade.js/blob/master/Dockerfile#L35) in the `Dockerfile` and it should work without issues (it did work during my tests under windows). I.e. the last few lines of your `Dockerfile` should look like this:
    ```Dockerfile
    ENTRYPOINT \
      source /emscripten/emsdk_env.sh &&\
      # python2.7 make.py && \
      python2.7 make.py wasm && \
      yarn && yarn generateTypes
    ```
    With that command, you are then only building the WASM-version of the library and not the JavaScript version. That means that there might be some older and/or mobile browsers which will not be compatible.

2. The other option you currently have (thanks to @zalo) is to use the automatic build system. Just fork the library and then push any changes changes to your fork. The build system will create the builds for you in ~20 minutes (it does not have the issues you reported).

I will try to do a bit more research on the topic (might also file an issue over at Emscripten) and let you know in case I find a fix for this.

donalffons, 2020-09-26:
...I upgraded the Emscripten version and made many changes to the build system since this issue was opened. The non-wasm version of the library has been deprecated now. Therefore, this issue should be solved. If you still experience problems, feel free to open a new one.

closed 2020-09-26 by donalffons