donalffons/opencascade.js #3: Typescript Definitions from the WebIDL

zalo, 2020-07-23:
[webidl2ts ](https://www.npmjs.com/package/webidl2ts) can be used to automatically generate typescript definitions from the `.idl` file here (ensure that the "-emscripten" mode is active).

This enables Intellisense completions in VS Code (and related editors)!

Here is an example generated typescript definitions file:
https://gist.github.com/zalo/0277def37489ef8c3ce47fc551aa243b

donalffons, 2020-07-23:
That sounds super cool! I will have a shot at using this tool and if it works well, I will add it to the project.

Long-term, I think it might be better to go with Emscripten's "Embind" workflow, instead of the "IDL" workflow (for [reasons](https://github.com/donalffons/opencascade.js/issues/1#issuecomment-660044413)... But I'm just guessing, haven't used Embind before). But for the moment, this will be super helpful to have in this library.

zalo, 2020-07-23:
Here's the use-case:
https://zalo.github.io/CascadeStudio/
I was inspired by your port and made a simple web-frontend to play with the API.

The typescript definitions are profoundly useful for discovering functionality while playing around (though it probably also makes sense to have a simplified "Standard Library" for more OpenSCAD like workflows...)

(By the way, could I trouble you to include `STEPControl_Writer` in the main API as well? 🙃 )

donalffons, 2020-07-23:
Wow, that just made my day. Yes sir, I'll get you the `STEPControl_Writer` stuff into the IDL + builds.
This tool looks super useful. I have also been thinking about publishing some examples using [CodeSandbox](https://codesandbox.io/) but didn't have a chance yet to work that yet. I like the super fast response you get when making changes with your tool though and might actually use this for prototyping, myself.

And yes, auto-completion and type checking is super useful when prototyping with the library and speeds up the development process a lot. I have only started using typescript a few months ago and don't want to miss it anymore.

How do you integrate the VS code editor into your website? That's such a helpful tool for building web applications...

zalo, 2020-07-24:
Thank you!  The VS Code editor is integrated via [monaco-editor](https://microsoft.github.io/monaco-editor/) (Microsoft's browser-friendly version of VS Code).

I'm hoping to expand the tool's features a lot (particularly save/load/import functionality) so that the BRep workflow can be a viable competitor to [OpenSCAD](https://www.openscad.org/) (and [OpenJSCAD](https://openjscad.org/)). :D

Also, not to be too much of a burden, but could you also consider including the other [Make Primitive Functions](https://www.opencascade.com/doc/occt-7.0.0/refman/html/package_brepprimapi.html)?  I had to do some unspeakable things to make a box earlier. 🙃

guidovanhilst, 2020-07-24:
> https://zalo.github.io/CascadeStudio/

Hey guys,  Really cool, nice work!

donalffons, 2020-07-24:
New builds are up!

donalffons, 2020-07-24:
@zalo, I also tried to run the `webidl2ts` script against the idl file with no success. I used this command
```
$ yarn webidl2ts -i ./opencascade.idl -n opencascade -ed -o ./dist/opencascade.d.ts
yarn run v1.22.4
$ /home/sebastian/Projects/opencascade.js/node_modules/.bin/webidl2ts -i ./opencascade.idl -n opencascade -ed -o ./dist/opencascade.d.ts
(node:173512) UnhandledPromiseRejectionWarning: WebIDLParseError: Syntax error at line 107, since `interface BRepPrimAPI_MakeCylinder`:
BRepPrimAPI_MakeCylinder implements BRepPrimAPI_MakeOneAxis;
^ Unrecognised tokens
    at tokeniser_Tokeniser.error (/home/sebastian/Projects/fiverr/2020-02-10_maxmanufacture_-_CAD_viewer/project/opencascade.js/node_modules/webidl2/dist/webidl2.js:1:20350)
    at r (/home/sebastian/Projects/fiverr/2020-02-10_maxmanufacture_-_CAD_viewer/project/opencascade.js/node_modules/webidl2/dist/webidl2.js:1:33487)
    at L (/home/sebastian/Projects/fiverr/2020-02-10_maxmanufacture_-_CAD_viewer/project/opencascade.js/node_modules/webidl2/dist/webidl2.js:1:34464)
    at Module.q (/home/sebastian/Projects/fiverr/2020-02-10_maxmanufacture_-_CAD_viewer/project/opencascade.js/node_modules/webidl2/dist/webidl2.js:1:34604)
    at Object.<anonymous> (/home/sebastian/Projects/fiverr/2020-02-10_maxmanufacture_-_CAD_viewer/project/opencascade.js/node_modules/webidl2ts/dist/parse-idl.js:47:43)
    at step (/home/sebastian/Projects/fiverr/2020-02-10_maxmanufacture_-_CAD_viewer/project/opencascade.js/node_modules/webidl2ts/dist/parse-idl.js:33:23)
    at Object.next (/home/sebastian/Projects/fiverr/2020-02-10_maxmanufacture_-_CAD_viewer/project/opencascade.js/node_modules/webidl2ts/dist/parse-idl.js:14:53)
    at /home/sebastian/Projects/fiverr/2020-02-10_maxmanufacture_-_CAD_viewer/project/opencascade.js/node_modules/webidl2ts/dist/parse-idl.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/home/sebastian/Projects/fiverr/2020-02-10_maxmanufacture_-_CAD_viewer/project/opencascade.js/node_modules/webidl2ts/dist/parse-idl.js:4:12)
(node:173512) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:173512) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Done in 0.87s.
```

It seems like the IDL parser has an issue with `BRepPrimAPI_MakeCylinder implements BRepPrimAPI_MakeOneAxis;` or any other statement of `A implements B;`. Did you have the same error? How did you manage to resolve that?

zalo, 2020-07-25:
Thank you for the example code!

I used a slightly different library and manually converted all of the implements to the colon inheritance syntax; however in the readme for this idl parser, it explicitly mentions the -e parameter as enabling a preprocess that does that conversion... maybe it’s not activating for some reason?

guidovanhilst, 2020-07-25:
I tried to make the "make bottle". (see code below)
But goes wrong due to override of oc.BRepBuilderAPI_MakeEdge.
Would it be nice if the make bottle worked?

I have made a C# script app to run OCC C# from the browser(backend) See also: [MakeBottle](http://www.creativecadtechnology.com/OCC/ShowScript?userName=occUser&groupName=demo&scriptName=MakeBottle)

```

function MakeBottle( myWidth,  myHeight,  myThickness)
        {
			var myNeckRadius = myThickness / 4;
            var myNeckHeight = myHeight / 10;
        
        
            let aPnt1 = new oc.gp_Pnt(-myWidth / 2, 0, 0);
            let aPnt2 = new oc.gp_Pnt(-myWidth / 2, -myThickness / 4, 0);
            let aPnt3 = new oc.gp_Pnt(0, -myThickness / 2, 0);
            let aPnt4 = new oc.gp_Pnt(myWidth / 2, -myThickness / 4, 0);
            let aPnt5 = new oc.gp_Pnt(myWidth / 2, 0, 0);

            //Profile : Define the Geometry
            let aArcOfCircle = new oc.GC_MakeArcOfCircle(aPnt2, aPnt3, aPnt4).Value();

            //Profile : Define the Topology
            let aEdge1 = new oc.BRepBuilderAPI_MakeEdge(aPnt1, aPnt2).Edge();
            let aEdge2 = new oc.BRepBuilderAPI_MakeEdge(aArcOfCircle).Edge();
            let aEdge3 = new oc.BRepBuilderAPI_MakeEdge(aPnt4, aPnt5).Edge();
            let aWire = new oc.BRepBuilderAPI_MakeWire(aEdge1, aEdge2, aEdge3).Wire();

            //Complete Profile
            let xAxis = oc.gp.OX();
            let aTrsf = new oc.gp_Trsf();

            aTrsf.SetMirror(xAxis);

            let aBRepTrsf = new oc.BRepBuilderAPI_Transform(aWire, aTrsf, false);
            let aMirroredShape = aBRepTrsf.Shape();
            let aMirroredWire = oc.TopoDS.Wire(aMirroredShape);

            let mkWire = new oc.BRepBuilderAPI_MakeWire();

            mkWire.Add(aWire);
            mkWire.Add(aMirroredWire);

            let myWireProfile = mkWire.Wire();

            //Body : Prism the Profile
            let myFaceProfile = new oc.BRepBuilderAPI_MakeFace(myWireProfile, true).Face();
            let aPrismVec = new oc.gp_Vec(0, 0, myHeight);

            let myBody = new oc.BRepPrimAPI_MakePrism(myFaceProfile, aPrismVec, true, false).Shape();

            //Body : Apply Fillets
            let mkFillet = new oc.BRepFilletAPI_MakeFillet(myBody, oc.ChFi3d_FilletShape.ChFi3d_Rational);
            let aEdgeExplorer = new oc.TopExp_Explorer(myBody, oc.TopAbs_ShapeEnum.TopAbs_EDGE, oc.TopAbs_ShapeEnum.TopAbs_SHAPE);

            while (aEdgeExplorer.More())
            {
                let aEdge = oc.TopoDS.Edge(aEdgeExplorer.Current());
                //Add edge to fillet algorithm
                mkFillet.Add(myThickness / 12, aEdge);
                aEdgeExplorer.Next();
            }

            myBody = mkFillet.Shape();

            //Body : Add the Neck	
            let neckLoc = new oc.gp_Pnt(0, 0, myHeight);
            let neckNormal = oc.gp.DZ();
            let neckAx2 = new oc.gp_Ax2(neckLoc.ation, neckNormal);
[... 141 lines cut ...]
    //cascadeViewport.updateShape(externalShapes['Pipe cut profile test Rhino step AP203.stp'], GUIState["Res"]);
	
	//Make bottle
    var Width=30.0;
    var Height=30.0;
    var  Thicknes=15.0;

    let bottle = MakeBottle(Width, Height, Thicknes);
    cascadeViewport.updateShape(bottle, GUIState["Res"]);
   
    console.log("Generation Complete!");
}

// Define GUI State Variables here
Object.assign(GUIState, {
    "Res"    : 0.1,  "ResRange"    : [0.01, 1.00],
    "Radius" : 30.0, "RadiusRange" : [27.0, 35.0] });

// GUI uses https://github.com/automat/controlkit.js
gui.addPanel({label: 'Cascade Control Panel'})
   .addButton     ('Generate Model' ,                             () => { CompileModel(); }  )
   .addSlider     (GUIState, 'Res',       'ResRange', { onFinish: () => { CompileModel(); }} )
   .addSlider     (GUIState, 'Radius', 'RadiusRange', { onFinish: () => { CompileModel(); }} );

```

zalo, 2020-07-25:
@guidovanhilst Perhaps better for another issue? (I feel bad about overloading this one so much already...)

MakeBottle (or a version of it) does appear to work, however; did you try the version from the example repo?
https://github.com/donalffons/opencascade.js-examples/blob/master/src/main.js#L46-L182

donalffons, 2020-07-25:
I found a super silly workaround for the problem with `webidl2ts` (will file an issue over there) and just added the type definition files in v0.1.16 this library. I tested it and it seems to work well. If you find any issues with it, let me know.

@guidovanhilst, I looked at your code. Unfortunately, one of the current limitations is that not all C++ methods can be exposed if they are overloaded by type of arguments only (vs by number of arguments). I randomly picked the following methods
```webidl
	void BRepBuilderAPI_MakeEdge([Const, Ref] Handle_Geom_Curve C);
	void BRepBuilderAPI_MakeEdge(
		[Const, Ref] Handle_Geom2d_Curve L,
		[Const, Ref] Handle_Geom_Surface S
	);
```
If you think, those methods should be changed to something else, let's start a new issue. I hope I will find a solution for this limitation soon.

@zalo, I agree - let's keep this thread from getting bigger and bigger.

Hope you don't mind if I will close this now.

closed 2020-07-25 by donalffons