donalffons/opencascade.js #9: how to correctly add new interface to opencascade.idl

iamnotstone, 2020-08-28:
I am trying to add a new interface to opencascade.idl like this:
```
interface BRepBuilderAPI_Copy{
  void BRepBuilderAPI_Copy();
  void BRepBuilderAPI_Copy(
    [Const, Ref]TopoDS_Shape,
    [Const]Standard_Boolean,
    [Const]Standard_Boolean
  );

  void Perform(
    [Const, Ref]TopoDS_Shape S,
    [Const]Standard_Boolean,
    [Const]Standard_Boolean
  );
};
```
And then, I run the" docker run" command .
Everything works fine, except I still didn't see a openCascade.BRepBuilderAPI_Copy() function in my javascript side.

Any one can tell me what I had missed ?

donalffons, 2020-08-28:
Hi,

in line 4 of your code snipped, it should say:
```webidl
    [Const, Ref]TopoDS_Shape S,
```
I.e., you're missing the **S** (or any other name for that parameter). Not including the **S**, gave an error during the `docker run` command. If you didn't see that error, this might mean that you're not correctly running the build(?!).

If you want to keep the default parameters, you could also add them, i.e.
```webidl
interface BRepBuilderAPI_Copy{
  void BRepBuilderAPI_Copy();
  void BRepBuilderAPI_Copy(
    [Const, Ref] TopoDS_Shape S,
    [Const] optional Standard_Boolean copyGeom = true,
    [Const] optional Standard_Boolean copyMesh = false
  );

  void Perform(
    [Const, Ref] TopoDS_Shape S,
    [Const] optional Standard_Boolean copyGeom = true,
    [Const] optional Standard_Boolean copyMesh = false
  );
};
```

I'm running a build with that change right now and will add it to the library once its done.

donalffons, 2020-08-28:
This is the [commit](https://github.com/donalffons/opencascade.js/commit/d498833cdffa949cd2345def8096ebefd37ea8df), if you want to check out which changes need to be made. Once the build is through, I will also update the npm package.

donalffons, 2020-08-28:
The new interface is now also available on npm as version 0.1.18.
By the way, maybe you did not run another `docker build` before your `docker run`?! Currently, you have to re-build the container each time using `docker build`, otherwise your changes to `opencascade.idl` will not be available during the build.

iamnotstone, 2020-08-29:
thank you very much! it worked ! I hadn't run the "docker build ..." command . 
So. If I want to add a new interface, I have to follow these steps:
1. to currectly add interface in opencascade.idl.
2. to add include file path in make.py if needed.
3. to create a new image by run the "docker build -t opencascde.js ." command and then run the "docker run ..." command.

closed 2020-09-01 by donalffons