donalffons/opencascade.js #21: Getting Face UVBounds

bitbybit-dev, 2021-01-26:
Hey guys, I'm trying to get UVBounds of the face. OCC documentation method is in occ.BRepTools.UVBounds_1.

https://dev.opencascade.org/doc/refman/html/class_b_rep_tools.html#a2223e42bb10997431eb16296ae740ea4

Method writes to passed Umin, Umax, Vmin and Vmax Standard_Real references, I'm a bit lost how to call this from javascript.

I tried smth like this:

```
        const uMin = new this.occ.TDataStd_Real();
        const uMax = new this.occ.TDataStd_Real();
        const vMin = new this.occ.TDataStd_Real();
        const vMax = new this.occ.TDataStd_Real();
        this.occ.BRepTools.UVBounds_1(face, uMin, uMax, vMin, vMax);
```

TypeError: Cannot convert "[object Object]" to double
I sort of get that, but yeah, **double** is not even a thing in javascript.

Anything else gives 
RuntimeError: call_indirect to a signature that does not match (evaluating 'cppInvokerFunc.apply(null,invokerFuncArgs)')

Any ideas?

donalffons, 2021-01-27:
Hi,

this is one of the limitations coming from the differences of how JavaScript and C++ work. The C++ function `BRepTools::UVBounds` takes in references to `Standard_Real`, which translates to a `Number` type in JavaScript. JavaScript does not support passing arguments as reference (only as shallow copy). I think this limitation only applies to built-in datatypes like float, double (or `Standard_Real`), int, char, etc.

In my understanding, there is currently no way to achieve this without modifying the OpenCascade API (with some Getter functions) and recompiling the library against this change. A change can be applied as a patch, so that there is a chance that it won't break with new releases of OpenCascade. There are already some similar patches in the 1.1.1 version of OpenCascade.js, like [this one](https://github.com/donalffons/opencascade.js/blob/v1.1.1/patches/Bnd_Box.hxx.patch).

I will open an issue in the Emscripten repository and ask if someone has a better workaround for this issue. If not, I will create the patch, which will then become part of the upcoming 2.0 release.

bitbybit-dev, 2021-01-27:
Thank you for the explanation and for looking into it! There's no urgency, and it's super nice that it will be addressed in 2.0. Also nice to know that there are ways to patch these things, I hope Emscripten team will have some ideas as well ;)

Your answer definitely saves me from banging against the wall. I suspected that there's something different about this one. Otherwise opencascade.js works amazing so far... Have to integrate few more basic features and will ship first experimental release.

donalffons, 2021-01-27:
Very happy to hear that :slightly_smiling_face:!
It would be great if you would share your results and issues, if you find more.

donalffons, 2021-06-09:
Have a look at [this post](https://github.com/donalffons/opencascade.js/issues/55#issuecomment-856025938) with an example on how to achieve this using a custom build.
As for the general solution to the problem of passing references to built-in datatypes: That will come eventually, but it might take a while.

closed 2021-06-09 by donalffons