donalffons/opencascade.js #30: Upgrade examples to use latest Webpack

donalffons, 2021-02-10:


bitbybit-dev, 2021-04-01:
It would be greatly appreciated if I could look into a working webpack example :) I've been trying to get v2.0 running in my app and custom webpack setup, but I'm having quite some problems with .d.ts files in opencascade.js/dist folder, they seem to have syntax errors. When I delete them I get some additional problems, probably related to es6 module loading, will be trying to continue experimenting. Also downloaded docker, maybe will try out to make a custom build. Not sure why file-loader would complain. Anyway, if it's not too hard to provide a running browser example for v2.0 it would be awesome

bitbybit-dev, 2021-04-02:
Ok, so I managed to solve this one ;)

donalffons, 2021-04-04:
How did you solve it? It would be appreciated if you could share a pull request, if you made some improvements.

bitbybit-dev, 2021-04-05:
Your instructions were correct, but I use angular and for custom webpack configs there's some additional setup required as standard angular does not expose webpack config, I thought that I was missing some context parameters... I'll be able to write a small guide on how to set up opencascade.js with angular if someone will want to use it that way, but at the end I only needed to add configuration that you provided and it worked. 

I may have used word 'solved' in a wrong way. I should have said 'made' it to run :) The real problem for others might be d.ts files which contain errors, or at least I think they do, cause if I open them up in vscode after doing npm_install they seem to contain unknown/non-standard types and if I start the app in debug mode all those errors get shown at runtime in the terminal. This I did not solve, just deleted those files from dist folder. This could also be ignored by skipLibCheck in tsconfig, but that is a global switch for all libraries so I did not use that one. And that is of course not a real 'solution' :) I'm not exactly sure how d.ts files get generated. Is it part of emscripten pipeline? Having looked at those d.ts files for a while I also couldn't really grasp what could be the root cause of those issues. 

Sometimes its simple stuff like in opencascade.d.ts
`export type openCascadeInstance = FS & {{ready: Promise<openCascadeInstance>}} & ...`
I suppose should be
`export type openCascadeInstance = FS & {ready: Promise<openCascadeInstance>} & ...`

This is the screenshot from /opencascade.js/dist/module.TKBinXCAF.wasm.d.ts which is missing BinMXCAFDoc_AssemblyItemRefDriver 

[image]

I guess you may understand better what is happening :) I was just happy that I could make it run and d.ts files were not necessary for me...

bitbybit-dev, 2021-04-05:
I'm really not sure if I'm up for a challenge like that, but if you would pinpoint the parts of the code that generate d.ts files I could look into it... ;) I'm also not really a python (or c++) guy, so can't make any promises if my help would turn out to be useful :D

donalffons, 2021-05-29:
Apologies for the late reply...
The `.d.ts` files are generated by opencascade.js' build system, not by Emscripten (unfortunately they don't support typescript for Embind).

I made several fixes to the `.d.ts` files and now they are "working" for me using Create-React-App's typescript config without major errors (i.e. without breaking the typescript compiler). However, there are still several issues with unknown types etc. which for me just show up as "any", since typescript cannot resolve all the types properly. I think it's a step in the right direction and I'm thinking this might be enough for finally releasing 2.0 - even if strictly speaking the types are still broken.

I've also upgraded the examples to use the latest webpack version `5.37.1` in [the beta branch](https://github.com/donalffons/opencascade.js-examples/tree/beta).

I'll work more on fixing the typescript definitions, but based on my tests, it should be useable now. If you still encounter major errors or issues, please reopen or post a new issue.

closed 2021-05-29 by donalffons

donalffons, 2021-05-29:
...sorry, my bad. I forgot to push the latest changes. As soon as [this build](https://github.com/donalffons/opencascade.js/runs/2700572387) his gone through, the mentioned typescript fixes should be part of the latest beta release.