donalffons/opencascade.js #33: GeomAPI_Interpolate seems to be missing from 1.1.4

bitbybit-dev, 2021-03-29:
Hi @donalffons, I hope you're doing good :) I was trying to use GeomAPI_Interpolate, but it seems like it is missing during the runtime, it is listed in Supported APIs.md file of 1.1.4 node_modules. Is there something special about this API? Would it appear if I would try out v2 beta?

bitbybit-dev, 2021-04-02:
Just managed to start v2.0. Loaded ocModelingAlgorithms, which includes TKGeomAlgo somehow also skips GeomAPI 
 together with GeomAPI_Interpolation. Actually by looking a bit deeper I think all packages from TKGeomAlgo are skipped. I've tried few approaches. Imported libs like this

```
initOpenCascade({
  libs: [
    ocCore,
    ocModelingAlgorithms,
    ocVisualApplication,
    ocDataExchangeBase,
    ocDataExchangeExtra,
  ]
}).then(oc => {
});
```

and spread out like this

```
initOpenCascade({
    libs: [
        ocCore,
        TKGeomAlgo, TKTopAlgo, TKShHealing, TKPrim, TKHLR, TKBO, TKMesh, TKBool, TKFillet, TKFeat, TKOffset,
        ocVisualApplication,
        ocDataExchangeBase,
        ocDataExchangeExtra
    ]
}).then(oc => {
});
```

But both of these seem to have the same problem. TKPrim apis are included fine.

donalffons, 2021-04-04:
Hi!
That API is currently blacklisted [here](https://github.com/donalffons/opencascade.js/blob/master/src/filter/filterClasses.py#L151), along with several other APIs, which caused all kinds of issues during build and/or startup of the WASM module.

I'm running a build right now and let you know what I find...

bitbybit-dev, 2021-04-05:
Thanks for that list, good to see excluded parts in one place together with reasons ;) This is not urgent, just wanted to create periodic closed curves and that seemed to be the easiest way. It's good to know that it wasn't skipped without a reason.

This also allowed me to try out v2.0 approach. Indeed having multiple modules takes longer to load, I also seem to experience some memory access issues on Chrome when running old scripts which worked with 1.1.4, Safari seems not to have those, but at this stage I think this may also be caused by 7.4.0 -> 7.5.0 transition or some other problems in my own code. Anyway, I am still experimenting with that and will provide feedback from my side a bit later.

donalffons, 2021-04-12:
I pushed a few commits last week which might solve this issue. First, I removed the previously mentioned blacklisting for those classes (including GeomAPI_Interpolate). The builds then caused errors during WASM startup due to some problematic methods. I then added blacklist entries for those methods ([here](https://github.com/donalffons/opencascade.js/blob/master/src/filter/filterMethods.py#L37)). The common thing among those methods seems to be that they have a declaration but no definition / implementation.

> This also allowed me to try out v2.0 approach. Indeed having multiple modules takes longer to load, I also seem to experience some memory access issues [...]

Yeah, to

bitbybit-dev, 2021-04-13:
Hey @donalffons, yeah, for me personally a single wasm file works really well, so I'm not against having few flavors with a 'full' and 'standard' versions. I suppose it will take some time to implement? Or is it available in latest beta?

The fact that there are such differences in wasm module loading times probably just means that it will take a while for emscript to really solve it, or maybe it is temporary, but fundamental wasm issue. I guess if you'll use this setup in development you'll notice if/when those shortcomings will be solved and then we will be able to switch.

donalffons, 2021-05-25:
I'm closing this now. If the problem still persists, feel free to re-open.

closed 2021-05-25 by donalffons

bitbybit-dev, 2021-09-10:
Hi @donalffons, had some time to update to the latest version - I see many improvements, also typescript works very well, and the interpolation API is now available, but it seems that for interpolation I need to construct TColgp_HArray1OfPnt, see [occt docs](https://dev.opencascade.org/doc/refman/html/class_geom_a_p_i___interpolate.html), but it doesn't seem to exist.

I've checked your [blacklist](https://github.com/donalffons/opencascade.js/blob/master/src/filter/filterMethods.py#L37) that you mentioned in above comments, but couldn't see that one. Is there a reason that this class is omitted? Maybe I can convert TColgp_Array1OfPnt to TColgp_Array1OfPnt? Thanks!

donalffons, 2021-09-10:
The bindings for these handles must (currently) be created at compile time. The build system currently creates bindings for all globally typedef'd handles (which include a lot, maybe most of those). But for the one mentioned by your issue, there is no such typedef.

I just [added](https://github.com/donalffons/opencascade.js/commit/de57295201b6fde4a8d69c2792064b444f1015a8) a typedef for `Handle_TColgp_HArray1OfPnt`. There will be a new beta release with that change in the next day or so. If you need it sooner, you can try a custom build :slightly_smiling_face: (with the yml file mentioned in that commit).

bitbybit-dev, 2021-09-10:
I get it. Thanks a lot for including that, I'm in no rush :) Will wait it out, cheers!

donalffons, 2021-09-10:
A new version `2.0.0-beta.bab6bec` is up, which contains the mentioned change. Would be great if you could try it and let me know if that does the trick!

bitbybit-dev, 2021-09-11:
Thanks for letting me know! I tried the new version and I see occ.Handle_TColgp_HArray1OfPnt constructors, but I sort of expected to see direct occ.TColgp_HArray1OfPnt constructors to initialize tables with points and some parameters. I'm probably quite a bit lost on the way those HArrays or OCC Tables need to be created... It seemed pretty straight forward with TColgp_Array1OfPnt, but it is not a table like structure.

It's also a bit strange that [Interpolation](https://dev.opencascade.org/doc/refman/html/class_geom_a_p_i___interpolate.html) docs do not have links anymore to TColgp_HArray1OfPnt. I guess I miss something.

If I construct the handle and try to get TColgp_Harray1OfPnt from there I get the unbound type error as well...

const d = new this.occ.Handle_TColgp_HArray1OfPnt_1();
const tcol = d.get();
---------------------
UnboundTypeError: Cannot call Handle_TColgp_HArray1OfPnt.get due to unbound types: P19TColgp_HArray1OfPnt

donalffons, 2021-09-11:
Thanks for testing that!
I also find some of the data structures of OpenCascade sometimes confusing...

I'll open a new issue regarding the missing TColgp_HArray1OfPnt class.

donalffons, 2021-09-11:
...it is probably going to take a while for me to investigate and resolve this. Keep in mind that with these blocking points, you might instead be able to write a small helper function (inside a helper class) as C++ code in the `additionalCppCode` block of a custom build. If you can make the interface of that helper function "simple" (without any of the problematic types), this might be a workaround (until I fix the underlying problem).