donalffons/opencascade.js #34: Consistently getting memory access out of bounds in v2.0

bitbybit-dev, 2021-04-09:
Hi, so I have this short snippet of code and it consistently gives me memory access out of bounds error on Chrome or crashes it completely with Aw Snap error. I've been getting that randomly in v1 with Difference commands and wasn't exactly sure where that comes from, but I hope I have it reduced to the mininum and reproducible. If that wouldn't be too hard to reproduce, maybe someone has any idea why would that particular code be problematic? Weirdly enough it does not crash when I have debugger open... Thanks.

`
        const pt1 = new oc.gp_Pnt_3(0, 0, 0);
        const x1 = new oc.BRepPrimAPI_MakeBox_3(pt1, 1, 2, 3).Shape();
        const pt2 = new oc.gp_Pnt_3(0, 1, 0);
        const x2 = new oc.BRepPrimAPI_MakeBox_3(pt2, 1, 1, 1).Shape();
        const differenceCut = new oc.BRepAlgoAPI_Cut_3(x1, x2);
        differenceCut.Build();
        const resShape = differenceCut.Shape();
`

donalffons, 2021-06-09:
Quick update: There are some indications that this might be a bug in Chrome stable (or to be more precise: V8's TurboFan optimizing compiler, which seems to be disabled when DevTools are open). There have been some [discussions over at Emscripten](https://github.com/emscripten-core/emscripten/issues/14391) about this problem and there is now [an issue on the V8 bug tracker](https://bugs.chromium.org/p/v8/issues/detail?id=11863) with a very similar behavior.
Crossing fingers that this gets resolved soon and actually fixes the issues we were experiencing here :crossed_fingers:.

bitbybit-dev, 2021-06-11:
Uff, that seems to go very deep indeed ;) Will be interesting to see the progress on these repos, thanks!

donalffons, 2021-06-12:
I tested your example code in latest versions of Chrome and Firefox several times and cannot reproduce the issue any longer. I think, it's fixed now!

closed 2021-06-12 by donalffons