donalffons/opencascade.js #35: Custom build with custom C++ classes

mnesarco, 2021-04-09:
First of all, thank you for this project.

Is it possible to use the docker container to create a custom build including some own c++ code?  I have created a c++ library using opencascade and I want to compile it to wasm using your opecascade port. But I don't see how to include my code in the custom build.

I don't need to expose anything from OpenCascade directly to javascript, I just need to expose some own c++ classes that uses openccascade internally, but I know very few about wasm/Emscripten. I suppose that if there is a way to inject my own code into the process, the custom build can build everything together and expose my classes as any other OCC class.

Thank you.

donalffons, 2021-04-12:
Hi!

> Is it possible to use the docker container to create a custom build including some own c++ code? I have created a c++ library using opencascade and I want to compile it to wasm using your opecascade port. But I don't see how to include my code in the custom build.

Hmm, that is an interesting use case, but not really something that is "supported" at the moment. Here are some ideas, though:

- Generally, you would probably have the best chances of success if your custom API follows the layout of the OpenCascade API, i.e. everything should be wrapped inside classes, no functions inside the global scope. You might also want to try to use a naming scheme, similar to OpenCascade...
- You might want treat your custom API as an OpenCascade "package" (i.e. similar to `math` or `gp`etc.). You should have all your source and header files inside a folder (no sub-folders) named `myApi`, for example.
- During the build process, a [Docker container](https://github.com/donalffons/opencascade.js/blob/master/Dockerfile#L114) is created which has the OpenCascade sources saved in `/occt/occt-94c0055/src`. You might be able to put your custom API in its own folder at this location using a `COPY myApi /occt/occt-94c0055/src` command inside the Dockerfile.
- You can create a custom build of the library ([doc](https://github.com/donalffons/opencascade.js/blob/master/doc/README.md#creating-a-custom-build), [example](https://github.com/donalffons/opencascade.js/blob/master/src/builds/customBuild.yml)). As your YAML file, you could use something like this
```yml
myApi.js:
  bindings:
    - package: "myApi"
    - module: "TKernel"
    - module: "TKMath"
    - module: "TKG2d"
    - module: "TKG3d"
    - module: "..." # Include all the modules you need
  
  emccFlags:
    - -sEXPORT_ES6=1
    - -sUSE_ES6_IMPORT_META=0
    - -sEXTRA_EXPORTED_RUNTIME_METHODS=["FS"]
    - -O3
    - -sAGGRESSIVE_VARIABLE_ELIMINATION=1
```

mnesarco, 2021-04-12:
Thank you @donalffons ,

Another possibility is to include my library similarly as you included freetype? Because refactoring it to mimic OCC structure can be very difficult at this time.

I will do some experiments.

Thank you again.

mnesarco, 2021-04-12:
Probably the solution is to combine two approaces:

1. To include the c++ code as a library (similar to freetype)
2. To Create a Facade mimic OCC structure just to export the desired functionality to javascript. as you described.

donalffons, 2021-05-28:
...there has been some progress with custom C++ code.
The custom builds (using the Docker image) now support a property named `additionalCppCode` in the YAML file. There is a note and an example on how to use it in [the docs](https://github.com/donalffons/opencascade.js/tree/beta/doc#creating-a-custom-build) for the beta version.

It is possible to add custom C++ code right from that YAML file. It should also be possible to #include additional C++ files from the folder, that is shared with the Docker container (but I haven't tested that yet). I would assume that this is enough for adding simple C++ helper classes to the code without having to fork the project.

mnesarco, 2021-05-28:
hi @donalffons it sounds great!

closed 2021-05-28 by donalffons

Vrainy, 2021-12-16:
> 
Is there any way to use "module"? It's more kindful than ”Symbol“.

I always missing relevant class when cofig symbol.