donalffons/opencascade.js #40: How to get the hierarchy of STP？

zjc0707, 2021-05-14:
I can only get one shape when use `STEPControl_Reader` to load a stp as in the opencascade-excample and CascadeStudio.
I found a [way](https://www.cnblogs.com/opencascade/p/XDE.html) to implement it in C++, but I can't do it in js
the code:
  ```
const fileText = await loadFileAsync(inputFile)
    const fileName = inputFile.name
    // Writes the uploaded file to Emscripten's Virtual Filesystem
    openCascade.FS.createDataFile('/', fileName, fileText, true, true)

    const aDoc = new openCascade.Handle_TDocStd_Document_1()
    console.log(aDoc)

    // XCAFApp is unsupported in API.md
    // const anApp = new openCascade.Handle_XCAFApp_Application_1().get()
    // console.log(anApp)

    const reader = new openCascade.STEPCAFControl_Reader_1()
    reader.SetColorMode(true)
    reader.SetNameMode(true)

    reader.ReadFile(fileName)

    console.log(reader)
    console.log(reader.NbRootsForTransfer())

    reader.Transfer_1(aDoc)

    const label = aDoc.get().Main()

    console.log(label)
```
the error in `reader.Transfer_1(aDoc)`

Is it the wrong way I use it, or is it not supported here ？

donalffons, 2021-05-14:
What error message do you get?
I believe you have to use the `Transfer` method like this (opencascade.js currently doesn't support default values for parameters):
```js
reader.Transfer_1(aDoc, new openCascade.Message_ProgressRange_1())
```

zjc0707, 2021-05-14:
I can't find `Message_ProgressRange_1` in .d.ts or Supported APIs.md

[image]
[image]

(I use the Version 1.1.1, but I found that the beta version has *. d.ts file, so I compare it)

donalffons, 2021-05-14:
Hmm, yeah, it seems like neither `Transfer_1` nor `Message_ProgressRange_1` is supported in OpenCascade.js 1.1.0. Could you try to update to the `beta` versions (`npm install opencascade.js@beta`) and follow the slightly changed instantiation procedure [explained in the README here on Github](https://github.com/donalffons/opencascade.js/tree/master/doc#using-pre-built-modules)? Also, keep in mind that the `beta` builds are using the latest OpenCascade Release 7.5.2 (e.g., if you want to check the OCCT docs).

zjc0707, 2021-05-14:
OK, I'll test it later and let you know the result
Thank you for your answer

zjc0707, 2021-05-17:
> Hmm, yeah, it seems like neither `Transfer_1` nor `Message_ProgressRange_1` is supported in OpenCascade.js 1.1.0. Could you try to update to the `beta` versions (`npm install opencascade.js@beta`) [...]

Sorry, I tried to change the version in opencascade.js-examples to beta, but I couldn't start the project. Do I need to modify webpack.config.js? Or can you provide an example of using the beta version?

[image]

error log:
```
ERROR in ./node_modules/opencascade.js/dist/module.TKVCAF.wasm
Module not found: Error: Can't resolve 'GOT.func' in '/Users/jcz/WebstormProjects/opencascade.js-examples/node_modules/opencascade.js/dist'
 @ ./node_modules/opencascade.js/dist/module.TKVCAF.wasm
 @ ./node_modules/opencascade.js/dist/index.js
 @ ./src/demos/bottle - basic/index.js

ERROR in ./node_modules/opencascade.js/dist/module.TKBinTObj.wasm
Module not found: Error: Can't resolve 'GOT.func' in '/Users/jcz/WebstormProjects/opencascade.js-examples/node_modules/opencascade.js/dist'
 @ ./node_modules/opencascade.js/dist/module.TKBinTObj.wasm
 @ ./node_modules/opencascade.js/dist/index.js
 @ ./src/demos/bottle - basic/index.js

ERROR in ./node_modules/opencascade.js/dist/module.TKTObj.wasm
Module not found: Error: Can't resolve 'GOT.func' in '/Users/jcz/WebstormProjects/opencascade.js-examples/node_modules/opencascade.js/dist'
 @ ./node_modules/opencascade.js/dist/module.TKTObj.wasm
 @ ./node_modules/opencascade.js/dist/index.js
 @ ./src/demos/bottle - basic/index.js

ERROR in ./node_modules/opencascade.js/dist/module.TKBinXCAF.wasm
Module not found: Error: Can't resolve 'GOT.func' in '/Users/jcz/WebstormProjects/opencascade.js-examples/node_modules/opencascade.js/dist'
 @ ./node_modules/opencascade.js/dist/module.TKBinXCAF.wasm
 @ ./node_modules/opencascade.js/dist/index.js
 @ ./src/demos/bottle - basic/index.js

ERROR in ./node_modules/opencascade.js/dist/module.TKBin.wasm
Module not found: Error: Can't resolve 'GOT.func' in '/Users/jcz/WebstormProjects/opencascade.js-examples/node_modules/opencascade.js/dist'
 @ ./node_modules/opencascade.js/dist/module.TKBin.wasm
 @ ./node_modules/opencascade.js/dist/index.js
 @ ./src/demos/bottle - basic/index.js

......

ERROR in ./node_modules/opencascade.js/dist/module.TKBO.wasm
WebAssembly module is included in initial chunk.
This is not allowed, because WebAssembly download and compilation must happen asynchronous.
Add an async splitpoint (i. e. import()) somewhere between your entrypoint and the WebAssembly module:
* ./src/demos/bottle - basic/index.js --> ./node_modules/opencascade.js/dist/index.js --> ./node_modules/opencascade.js/dist/module.TKBO.wasm
* multi (webpack)-dev-server/client?http://localhost:9000 ./src/demos/bottle - basic/index.js --> ./src/demos/bottle - basic/index.js --> ./node_modules/opencascade.js/dist/index.js --> ./node_modules/opencascade.js/dist/module.TKBO.wasm

ERROR in ./node_modules/opencascade.js/dist/module.TKBRep.wasm
WebAssembly module is included in initial chunk.
This is not allowed, because WebAssembly download and compilation must happen asynchronous.
Add an async splitpoint (i. e. import()) somewhere between your entrypoint and the WebAssembly module:
* ./src/demos/bottle - basic/index.js --> ./node_modules/opencascade.js/dist/index.js --> ./node_modules/opencascade.js/dist/module.TKBRep.wasm
* multi (webpack)-dev-server/client?http://localhost:9000 ./src/demos/bottle - basic/index.js --> ./src/demos/bottle - basic/index.js --> ./node_modules/opencascade.js/dist/index.js --> ./node_modules/opencascade.js/dist/module.TKBRep.wasm

ERROR in ./node_modules/opencascade.js/dist/module.TKBin.wasm
WebAssembly module is included in initial chunk.
This is not allowed, because WebAssembly download and compilation must happen asynchronous.
Add an async splitpoint (i. e. import()) somewhere between your entrypoint and the WebAssembly module:
* ./src/demos/bottle - basic/index.js --> ./node_modules/opencascade.js/dist/index.js --> ./node_modules/opencascade.js/dist/module.TKBin.wasm
* multi (webpack)-dev-server/client?http://localhost:9000 ./src/demos/bottle - basic/index.js --> ./src/demos/bottle - basic/index.js --> ./node_modules/opencascade.js/dist/index.js --> ./node_modules/opencascade.js/dist/module.TKBin.wasm

ERROR in ./node_modules/opencascade.js/dist/module.TKBinL.wasm
WebAssembly module is included in initial chunk.
This is not allowed, because WebAssembly download and compilation must happen asynchronous.
Add an async splitpoint (i. e. import()) somewhere between your entrypoint and the WebAssembly module:
* ./src/demos/bottle - basic/index.js --> ./node_modules/opencascade.js/dist/index.js --> ./node_modules/opencascade.js/dist/module.TKBinL.wasm
* multi (webpack)-dev-server/client?http://localhost:9000 ./src/demos/bottle - basic/index.js --> ./src/demos/bottle - basic/index.js --> ./node_modules/opencascade.js/dist/index.js --> ./node_modules/opencascade.js/dist/module.TKBinL.wasm

.....

Child HtmlWebpackCompiler:
     3 assets
    Entrypoint HtmlWebpackPlugin_0 = __child-HtmlWebpackPlugin_0
    Entrypoint HtmlWebpackPlugin_1 = __child-HtmlWebpackPlugin_1
    Entrypoint HtmlWebpackPlugin_2 = __child-HtmlWebpackPlugin_2
    [./node_modules/html-webpack-plugin/lib/loader.js!./src/demos/bottle - basic/index.html] 1.41 KiB {HtmlWebpackPlugin_0} [built]
    [./node_modules/html-webpack-plugin/lib/loader.js!./src/demos/bottle - visualize/index.html] 1.41 KiB {HtmlWebpackPlugin_1} [built]
    [./node_modules/html-webpack-plugin/lib/loader.js!./src/demos/polygon/index.html] 625 bytes {HtmlWebpackPlugin_2} [built]
ℹ ｢wdm｣: Failed to compile.

```

donalffons, 2021-05-17:
I just created a new [`beta` branch in the examples repository](https://github.com/donalffons/opencascade.js-examples/commit/d0ebdb5f9d2ef456da76482e681de073a4e9b470).

Fixing these issues is mostly just a matter of tweaking the Webpack config.
[image]

Currently, IGES support seems to be broken, probably due to an Emscripten issue (see [here](https://github.com/emscripten-core/emscripten/issues/13219)).

zjc0707, 2021-05-18:
Thank you very much. Now I can use the beta version. Then I tested my code and it seemed that there was a new error about `reader.Transfer_1(aDoc, new openCascade.Message_ProgressRange_1())`. Can you give some suggestions on using `STEPCAFControl_Reader_1`? Or is there another way to decompose multiple shapes from a step file?

```
    const aDoc = new openCascade.Handle_TDocStd_Document_1()
    console.log(aDoc)

    // XCAFApp is unsupported in API.md
    // const anApp = new openCascade.Handle_XCAFApp_Application_1().get()
    // console.log(anApp)

    const reader = new openCascade.STEPCAFControl_Reader_1()
    reader.SetColorMode(true)
    reader.SetNameMode(true)

    reader.ReadFile(fileName)

    console.log(reader)
    console.log(reader.NbRootsForTransfer())

    reader.Transfer_1(aDoc, new openCascade.Message_ProgressRange_1())

    console.log(reader.NbRootsForTransfer())

    const label = aDoc.get().Main()

    console.log('label')
    console.log(label)
```

[image]

zjc0707, 2021-05-18:
I'm going to parse the file structure directly through the content of the step file, and then use opencascade.js to load it. At present, it seems feasible. Finally, thank you very much for your work in this area

donalffons, 2021-05-18:
I think the reason for that exception is in your first line:
You create a `Handle` for a `TDocStd_Document`. I think the `Handle` class works in C++ like a memory-managed smart pointer. In this state, `aDoc` does not have any memory assigned to it. To create the handle and assign memory to it, try the following:
```js
const aDoc = new openCascade.Handle_TDocStd_Document_2(new openCascade.TDocStd_Document(new openCascade.TCollection_ExtendedString_1()));
```
the extra parameters are just the default parameters. Please let me know if that fixes your problem.

donalffons, 2021-05-18:
...as for analyzing the hierarchy of a STEP file: I haven't done that yet (but would also really like to understand how that works). There are some hints in the OpenCascade Forum, for example [in this thread](https://dev.opencascade.org/content/load-step-file-hierarchy-tree).

donalffons, 2021-05-18:
...one more thing: If you want to show the 3D model using something like ThreeJS with colors, take a look at `RWGltf_CafWriter`. With that you could export your tessellation into a GLB file, which could in turn be loaded using ThreeJS (that way you could avoid the complex tessellation code from the "Helper" library in the examples).

zjc0707, 2021-05-18:
> I think the reason for that exception is in your first line:
> You create a `Handle` for a `TDocStd_Document`. I think the `Handle` class works in C++ like a memory-managed smart pointer. In this [...]

After this modification, `transfer_1` can be used. I will continue to try the operation on the `label`. Thank you very much.

zjc0707, 2021-05-18:
> ...as for analyzing the hierarchy of a STEP file: I haven't done that yet (but would also really like to understand how that works). There are some hints in the OpenCascade Forum, for example [in this thread](https://dev.opencascade.org/content/load-step-file-hierarchy-tree).

In the step file, I find that `PRODUCT` defines each component, and `NEXT_ASSEMBLY_USAGE_OCCURRENCE` defines the parent-child relationship between products.
So, I just need to find out the content between the `PRODUCT` and the next `PRODUCT`, and make up the other referenced data (# + ID), and then I can split the step file.

show my code
```js

export async function testStep(file: File) {
  const header = 'ISO-10303-21;\n' +
    'HEADER;\n' +
    'FILE_DESCRIPTION((\'STEP AP214\'),\'1\');\n' +
    'FILE_NAME(\'test.stp\',\'2020-10-30T08:18:32\',(\' \'),(\' \'),\'Spatial InterOp 3D\',\' \',\' \');\n' +
    'FILE_SCHEMA((\'AUTOMOTIVE_DESIGN { 1 0 10303 214 1 1 1 1 }\'));\n' +
    'ENDSEC;\n\n' +
    'DATA;\n'

  const footer = '\nENDSEC;\n' +
    'END-ISO-10303-21;\n\n'

  const productIndexes: number[] = []
  const arr: (string | undefined)[] = []
  const fileText = (await ReaderService.Instance.readFile(file)).replace(/\r/g, '')
  fileText.split('\n')
    .filter(p => p.startsWith('#'))
    .forEach(p => {
      const index = Number.parseInt(p.substring(1, p.indexOf('=')))
      arr[index] = p
      if (p.includes('PRODUCT(')) {
        productIndexes.push(index)
      }
    })

  console.log(arr.length)

  const files: File[] = []
  for (let i = 0; i < productIndexes.length; i++) {
    const start = productIndexes[i]
    const end = ((i + 1 < productIndexes.length) ? productIndexes[i + 1] : arr.length)
    console.log(end)

    const stack: (string | undefined)[] = []
    const rs: (string | undefined)[] = []
    for (let j = start; j < end; j++) {
      stack.push(arr[j])
      rs[j] = arr[j]
    }

    while (stack.length > 0) {
      const s = stack.shift()?.substring(1)
      if (!s || !s.includes('#')) {
        continue
      }

      s.split('#')
        .forEach(p => {
          let b = p.length
          for (let j = 0; j < p.length; j++) {
            const code = p.charCodeAt(j)
            if (code < 48 || code > 57) {
              b = j
              break
            }
          }
          const num = Number.parseInt(p.substring(0, b))
          if (isNaN(num)) {
            console.error('nan')
          }
          if (!rs[num]) {
            stack.push(arr[num])
            rs[num] = arr[num]
          }
        })
    }

    const title = arr[start]!
    let filename = title.substring(title.indexOf('('), title.indexOf(','))
    filename = filename.substring(filename.indexOf('\'') + 1, filename.lastIndexOf('\''))
    if (rs.some(p => p && p.includes('CARTESIAN_POINT('))) {
      console.log(filename)
      const s = header + rs.filter(p => p).join('\n') + footer
      files.push(LoadSaveFile.getFile(s, `${filename}.stp`))
    }

    // console.log(s)
    // FileSaver.saveAs(LoadSaveFile.getBlob(s), `${filename}.stp`)
  }

  console.log(files)
}

```

So far, I've only done a simple test and it seems to work. I haven't dealt with the parent-child hierarchy yet, so there will be step files that can't be loaded because they are just an empty group.

[Front_Assembly.step.zip](https://github.com/donalffons/opencascade.js/files/6499182/Front_Assembly.step.zip)
This is the file I used for testing, it has one parent product and eight sub products

zjc0707, 2021-05-18:
> ...one more thing: If you want to show the 3D model using something like ThreeJS with colors, take a look at `RWGltf_CafWriter`. With that you could export your tessellation into a GLB file, which could in turn be loaded using ThreeJS (that way you could avoid the complex tessellation code from the "Helper" library in the examples).
> EDIT: I have posted an example on how to do this a while ago [here](https://github.com/donalffons/opencascadejs-tessellation-test/blob/master/src/opencascade.worker.ts). Code might be a bit outdated now, but in general this approach should still work.
> EDIT2: [Preview link](https://opencascadejs-tessellation-test.donalffons.vercel.app/)

Thank you very much. I'll try this

donalffons, 2021-05-28:
I had another look at this today. Here are some more bits and pieces of information. By the way, feel free to ignore this if you managed to solve your problem with your custom code.

So, eventually, we want to get the names of the items in a STEP file. Currently, there is no way convert any of OpenCascade's strings or simple C-style string to a JavaScript string (I added a [feature request](https://github.com/emscripten-core/emscripten/issues/14314) over at Emscripten, but not sure if that's something they're planning to support). I added a new class called `Helpers` (in the `beta` branch) which has a static function `ConvertAsciiString` that converts a `TCollection_AsciiString` into a JavaScript supported string.

With that added function, you can use this code to iterate over children of a shape and print the respective `name` label.

```js
    const assembly = this.oc.XCAFDoc_DocumentTool.ShapeTool(this.doc.get().Main());
    const freeShapes = new this.oc.TDF_LabelSequence_1();
    assembly.get().GetFreeShapes(freeShapes);
    this.oneFreeShape = freeShapes.First();
    for(let it = new this.oc.TDF_ChildIterator_2(this.oneFreeShape, true); it.More(); it.Next()) {
      const att = new this.oc.Handle_TDF_Attribute_2(new this.oc.TDataStd_Name());
      it.Value().FindAttribute_1(this.oc.TDataStd_Name.GetID(), att);
      const name = this.oc.Helpers.ConvertAsciiString(new this.oc.TCollection_AsciiString_13(att.get().Get(), 0));
      console.log(name);
    }
```

I also found this OpenCascade forum post, which seems to have a more complete solution - seemingly using `XCAFDoc_DocumentTool::ShapeTool`.

donalffons, 2021-05-29:
This code should do the trick. It traverses the assembly and saves all names and assemblies in a tree structure. There are some typescript annotations in there.
```js
    type Assembly = {
      name: string;
      children: (Assembly | Part)[];
    };

    type Part = {
      name: string;
    };

    const traverseLabel: (oc: openCascadeInstance, label: TDF_Label) => (Assembly | Part) = (oc, label) => {
      const att = new oc.Handle_TDF_Attribute_2(new oc.TDataStd_Name());
      label.FindAttribute_1(oc.TDataStd_Name.GetID(), att);
      // @ts-ignore
      const name = oc.Helpers.ConvertAsciiString(new oc.TCollection_AsciiString_13((att.get() as TDataStd_Name).Get(), 0));

      const components = new oc.TDF_LabelSequence_1();
      if(oc.XCAFDoc_ShapeTool.GetComponents(label, components, false)) {
        const children: (Assembly | Part)[] = [];
        for(let i = 1; i <= components.Length(); i++) {
          children.push(traverseLabel(oc, components.Value(i)));
        }
        return {
          name,
          children,
        }
      } else {
        const referredShapeLabel = new oc.TDF_Label();
        if(oc.XCAFDoc_ShapeTool.GetReferredShape(label, referredShapeLabel)) {
          return traverseLabel(oc, referredShapeLabel);
        } else {
          return { name };
        }
      }
    }

    const traverseAssembly = (oc: openCascadeInstance, assembly: Handle_XCAFDoc_ShapeTool) => {
      const freeShapes = new oc.TDF_LabelSequence_1();
      assembly.get().GetFreeShapes(freeShapes);
      const children = [];
      for(let i = 1; i <= freeShapes.Length(); i++) {
        children.push(traverseLabel(oc, freeShapes.Value(i)));
      }
      return children;
    }
    // this.doc is of type Handle_TDocStd_Document
    if(!this.doc) throw new Error();
    console.log(traverseAssembly(this.oc, this.oc.XCAFDoc_DocumentTool.ShapeTool(this.doc.get().Main())));
```
[image]

closed 2021-05-29 by donalffons

zjc0707, 2021-05-31:
Thank you very much. I will refer to it
I'm very glad to have your help. Good luck

bitbybit-dev, 2022-12-16:
Hey @donalffons, hope you're doing well. Thanks for this snippet, super helpful. I was looking for ConvertAsciiString and Helpers class, but could not find that on beta branch. Could you point me to where it is? I thought it should be on additionalCppCode on yml build? Thanks! ;)

ranaraumik16, 2023-06-08:
Hello @donalffons ,
I tried to get names using above guidance, but I cannot find "oc.Helpers".
I am using "2.0.0-beta.b5ff984" version. Please help me with this.

domenicosmartup, 2025-05-14:
> Hello [@donalffons](https://github.com/donalffons) , I tried to get names using above guidance, but I cannot find "oc.Helpers". I am using "2.0.0-beta.b5ff984" version. Please help me with this.

Did you find a solution?