donalffons/opencascade.js #41: Unable to load slightly complex STP in opencascade-example

zjc0707, 2021-05-14:
[image]

I have many STP files, but none of them can be loaded except box.
But I can load all of them in CascadeStudio.

I find the api is different, how can I use it ?

donalffons, 2021-05-14:
Could you share one of the problematic STEP files?

zjc0707, 2021-05-14:
[step.txt](https://github.com/donalffons/opencascade.js/files/6477470/step.txt)

I don't seem to be able to upload step files, so I modified the suffix

donalffons, 2021-05-14:
Hmm,

for me this part is working in the [examples / demo](https://donalffons.github.io/opencascade.js-examples/demos/bottle%20-%20basic/).

[image]

I tried it in Chrome + Firefox with OpenCascade.js version 1.1.1 and the current beta versions.

I saw this type of error before (but wasn't able to fix it then) - appreciate your help with this. Any tips on how I can reproduce this?

zjc0707, 2021-05-14:
[error.zip](https://github.com/donalffons/opencascade.js/files/6478363/error.zip)

Thank you for your reply

I have tried many times and found that as long as the file name has been changed (in MacOS), the non loadable files become loadable. The file in this package is the same as the previous file, but it cannot be loaded without changing the file. Please test it

And after I refresh the page many times, the following situation will appear, and then I have to restart the browser
[image]

zjc0707, 2021-05-14:
In addition, this STP file is exported from Solidworks on windows. But now I don't have a Windows computer around me, so I can't test the result of modifying this file under windows. I hope it can help

donalffons, 2021-05-14:
> I have tried many times and found that as long as the file name has been changed (in MacOS), the non loadable files become loadable. The file in this package is the same as the previous file, but it cannot be loaded without changing the file. Please test it

Thanks a lot for that! I was able to reproduce the exact same behavior on my machine. It seems like there is something with that filename that OpenCascade (or Emscripten's file system emulator?) doesn't like. I changed the code in the examples repository slightly to [rename the file](https://github.com/donalffons/opencascade.js-examples/blob/master/src/demos/bottle%20-%20basic/library.js#L40) before saving it to the virtual file system. That seems to fix the issue. I've updated the example deployment - feel free to try.

zjc0707, 2021-05-14:
Yes, I can use it when I modify the `fileName` in code. This is really a strange question。
thanks

donalffons, 2021-05-14:
...and regarding the "out of memory" error: Yeah, that's kind of expected (unfortunately), as the example code currently doesn't do a great job of cleaning up used memory. Memory cleanup has to be done all manually at the moment. For each `new`, there has to be a `delete` as discussed [here](https://github.com/donalffons/opencascade.js-examples/issues/8). (Unfortunately, this doesn't help with code readability.)

Having to restart the browser seems quite odd, though... I will try to reproduce this and might ask over at Emscripten or Firefox if this is intended behavior.

zjc0707, 2021-05-14:
It's OK to restart Firefox, Google browser will go down directly, and then it can only be shut down through the activity monitor. Maybe this is a problem with my computer. After all, the adaptation of M1 is still in progress

zjc0707, 2021-05-14:
> ...and regarding the "out of memory" error: Yeah, that's kind of expected (unfortunately), as the example code currently doesn't do a great job of cleaning up used memory. Memory cleanup has to be [...]

It's my fault. Refreshing the page is still wrong. Just create a new browser page. I don't need to restart the whole browser. But I think most people's first reaction should be to constantly refresh the page, especially when browsing online pages

donalffons, 2021-05-14:
Which version of MacOS are you using? Big Sur?

zjc0707, 2021-05-14:
> Which version of MacOS are you using? Big Sur?

yes

closed 2021-05-18 by zjc0707