donalffons/opencascade.js #45: An error occurred using the latest version,

WREQI, 2021-05-25:
I installed and configured according to the document https://github.com/donalffons/opencascade.js/tree/master/doc#using-pre-built-modules, but received the following error. What could be the reason?

[image]

donalffons, 2021-05-25:
Hmm, that probably means that you don't have the latest version of the `opencascade.js@beta` package installed (I accidentally published one or more broken versions some time ago).

If you open up the file `./node_modules/opencascade.js/package.json` in your local project, it should say `"version": "2.0.0-beta.bf4d306"` in line 3. I assume that it has a different version there. To fix this, you need to:

1. run `yarn remove opencascade.js` (or npm)
2. delete your `node_modules` folder
3. delete your `package-lock.json` file (if you're using npm) or your `yarn.lock` file (if you're using yarn)
4. run `yarn add opencascade.js@beta`

It seems that the problem is that the beta builds are not versioned semantically, which gets yarn/npm confused. If you're using the examples, try to `git pull` that project as I pushed a fix for that issue today.

donalffons, 2021-05-25:
...also: You have to do these steps every time you want to update the `opencascade.js@beta` version.

I will hopefully be able to publish a 2.0 version soon - that problem should go away then.

closed 2021-05-25 by WREQI