donalffons/opencascade.js #46: how to build .wasm file for beta version

iamnotstone, 2021-05-26:
Hello. I am trying to build the wasm myself. first I run "docker build -t opencascade.js ." and then I run "docker run -ti --rm opencascade.js"  It seems works fine but I didn't see any .wasm files ?

donalffons, 2021-05-26:
Hey,
the reason why you don't see any output files is that the docker container cannot just access your file system without you giving it permission to do so. There are [instructions in the docs](https://github.com/donalffons/opencascade.js/tree/master/doc) (at the very bottom) on how to share your current working directory with the docker container and also how to set the correct user for the output files (on linux - not sure if that does anything on windows).

If you want to build your own custom version, follow the steps described in the docs. If you want to re-build any of the builds from the beta branch, have a look at the github workflows:

- [Standard modules](https://github.com/donalffons/opencascade.js/blob/c3f87cca09a6f356881e577b6d1cb5518f2c7086/.github/workflows/buildDefault.yml#L31): OpenCascade-defined packages, e.g. TKernel, TKMath, TKBRep, ...
- [Combined modules](https://github.com/donalffons/opencascade.js/blob/c3f87cca09a6f356881e577b6d1cb5518f2c7086/.github/workflows/buildDefault.yml#L41): Pre-built combined modules [as described in the docs](https://github.com/donalffons/opencascade.js/tree/master/doc#types-of-pre-built-modules).
- [Full version](https://github.com/donalffons/opencascade.js/blob/c3f87cca09a6f356881e577b6d1cb5518f2c7086/.github/workflows/buildFull.yml#L31): Everything in one big chunk, currently not part of the beta packages

Basically, you have to add a `-v` argument (and probably a `-u` also) to your `docker run` command.

iamnotstone, 2021-05-28:
thanks!  it solved my problem.

closed 2021-05-28 by donalffons