donalffons/opencascade.js #54: How to save the wasm file into browser cache, such as IndexedDB?

zjc0707, 2021-06-04:
I checked the [documentation](https://developer.mozilla.org/en-US/docs/WebAssembly/Caching_modules), but it loads wasm directly and uses the instance of the result. In opencascade.js, the processing of wasm seems to be integrated into opencascade(). How can I cache?

donalffons, 2021-06-04:
Probably shouldn't be trying to use IndexedDB caching, as this is a feature which is being removed from browsers (see the warning at the top of the page that you linked). Instead, you probably want to leave caching to the browser and set appropriate headers on your server.

Sadly, the modularized version of OpenCascade.js cannot be properly cached by browsers at the moment, since the dynamic modules approach requires a lot of "dynamic" JS execution when loading modules. I would encourage you (and should probably mention that in the docs) to use the npm package of OpenCascade.js mainly for development purposes (as you have the full API available and can make experiments quickly). If you want to ship a production app, you probably want to create your own custom build without any dynamically loaded modules. This will then start up much faster.

Here is a small [STEP file loader ](https://interactive-cad-tool.web.app/)that I have been working on. It uses a custom build and is ready to start (that is, when the `loading` message goes away) after after ~2-3 seconds on my laptop on the first visit. On subsequent visits, it is ready pretty much instantly.

For OpenCascade.js, there would be a better alternative than using the dynamic modules, which might be better to use during development: Using a single WASM file with the "full" version of the library. This would load quite a bit faster. I documented some benchmarks, [here](https://github.com/emscripten-core/emscripten/issues/13495). However, the problem with this approach is that the build times are currently extremely long (> 30 hours) which would make it extremely hard for me to keep improving the library. I recently had a conversation with one of the Emscripten devs, who gave me some tips on how to significantly speed this up. I am going to implement these changes, but that's going to take some time.

zjc0707, 2021-06-04:
Thanks, I used the full WASM file from version 1.1.1. Now I can cache by setting up nginx.
Finally, the [app](https://interactive-cad-tool.web.app/) is so cool, can you share the code?

donalffons, 2021-06-04:
I will share some parts of that app as part of the examples (in plain JS, i.e. without React), most importantly:

- provide a working example for how to create custom builds
- show how to use `StepCAFControl_Reader`, `BRepMesh_IncrementalMesh` for tessellation and `RWGLTF_CAFWriter` and how to import the resulting GLTF file into ThreeJS
- show how to enable multi-threading for a build and use that for tessellation

However, I don't think I want to make that specific app completely open source just yet.

But if you have any specific questions feel free to ask. I'm happy to help :slightly_smiling_face:.

closed 2021-06-04 by donalffons