donalffons/opencascade.js #57: Make viewport responsive in examples

donalffons, 2021-06-12:


theRealSuperMario, 2021-09-09:
Hi,

I was checking out the examples and also noticed that the examples a quite a big laggy and have a high memory consumption. Am I getting this right that this issue is documenting exactly that?

If so, I guess there is no need to create a new issue.

Thanks for the great repo 🙏

donalffons, 2021-09-09:
Hi,

what I originally meant with this issue was to just make the 3D viewport resize when you resize the browser window. Nothing too fancy :slightly_smiling_face:.

When you say that the examples are laggy, you mean that your browser cannot sustain a constant (high) frame rate while you're using the sliders?!
There are probably 2 sides to that problem:
1. All code of the examples currently run on the main thread. In a production application, you would probably want to put all the compute-intensive OpenCascade.js code into a web worker, which runs in its own separate thread and is therefore not blocking the browsers main / ui thread.
2. There might also be some potential to improve the runtime performance of the OpenCascade.js build of the examples. From the top of my head, here are some hints: Enable Emscripten's link-time-optimizations, use multi-threading, improve tessellation code.

Regarding the memory consumption: Each example is currently loading the entire OpenCascade library, which contains large amounts of unnecessary code for each particular example. This results in the high memory consumption (and potentially sub-optimal performance).
For a production app, you would probably want to create a custom build, which contains only the parts of the API that you specify. Maybe some day, when WASM modules are standardized (and have reasonably good performance), this won't be necessary any more - but at the moment it is.

I would like to keep the existing examples as simple as possible (i.e. not implement these optimizations). However, there is certainly room for additional examples and/or docs...

theRealSuperMario, 2021-09-10:
Thanks for the clarification.

This sounds totally reasonable.

donalffons, 2022-03-29:
The examples will be re-written to use Google's `<model-viewer>` component in the near future. There is no need to put more effort into this issue.

closed 2022-03-29 by donalffons