donalffons/opencascade.js #59: yarn build error

WREQI, 2021-06-15:
[image]
1. yarn start error
2. yarn build error

donalffons, 2021-06-15:
1. `yarn remove opencascade.js`
2. `yarn add opencascade.js@beta`

should fix that. This will install the latest version from NPM.

If that doesn't work, please share the `version` that is defined in `./node_modules/opencascade.js/package.json`.

WREQI, 2021-06-15:
[image]
[image]
Can start, build error

donalffons, 2021-06-15:
Do you get a different error message now?
Because `WNT_HIDSpaceMouse` should not be included in your version of OpenCascade.js, see the [typescript definitions here](https://unpkg.com/browse/opencascade.js@2.0.0-beta.e96854d/dist/module.TKService.wasm.d.ts).

WREQI, 2021-06-15:
[image]
Looks like there is a syntax error somewhere that does not support strict mode?

donalffons, 2021-06-15:
Hmm, right... Maybe there is an issue with the typescript definition.
I'm having a Create-React-App open which also uses in <React.StrictMode> and which imports OpenCascade.js into `App.tsx`. I can call `initOpenCascade` from that file without issues. 

Could you maybe share a some code or more information on you configuration?

WREQI, 2021-06-17:
> 
[config-overrides.txt](https://github.com/donalffons/opencascade.js/files/6668337/config-overrides.txt)
This is my configuration file. Since the js file cannot be uploaded, I changed it to txt upload

WREQI, 2021-06-17:
Can you upload your react demo to github? What is the difference in version configuration under our download and run comparison?

WREQI, 2021-06-18:
> 嗯，对...也许打字稿定义有问题。
> 我打开了一个 Create-React-App，它也使用 <React.StrictMode> 并将 OpenCascade.js 导入到`App.tsx`. 我可以`initOpenCascade`毫无问题地从该文件中调用。
> 
> 您能否分享一些有关您的配置的代码或更多信息？

Can you take a look at your Create-React-App?

donalffons, 2021-06-18:
Hmm, sorry, I cannot share that project here.
Here is roughly the minimum that you need to create and configure Create-React-App to use with OpenCascade.js@beta

1. Create the app `yarn create react-app my-app`
2. Add opencascade.js, i.e. `yarn add opencascade.js@beta`
3. Add and configure `react-app-rewired`
4. Create a `config-overrides.js`, like this
```js
module.exports = {
  webpack: (config) => {
    config.module.rules.find(k => k.oneOf !== undefined).oneOf.unshift(
      {
        test: /\.wasm$/,
        type: "javascript/auto",
        loader: "file-loader",
        options: {
          name: "static/js/[name].[contenthash:8].[ext]",
        },
      }
    );
    return config;
  },
};
```
...and then just use it like you would normally (see the [examples](https://github.com/donalffons/opencascade.js-examples)). For me that workflow works just fine.

closed 2021-06-21 by WREQI