donalffons/opencascade.js #60: "UnboundTypeError" for custom builds

sgenoud, 2021-06-17:
I was trying to create a custom build - the build worked, but there is an issue with the build result. I have the following message in the console:

```
"Cannot call gp_Vec.XYZ due to unbound types: 6gp_XYZ"
```

From what I understand you are adding already adding "raw pointers" in order to avoid this kind of error. What am I missing?

My custom build setup is the following: 

```.yml
cadeau_single.js:
  bindings:
    - symbol: Message_ProgressRange
    - symbol: TColgp_Array1OfDir

    - symbol: gp_XYZ
    - symbol: gp_Vec
    - symbol: gp_Pnt
    - symbol: gp_Dir
    - symbol: gp_Ax2
    - symbol: gp_Ax3
    - symbol: gp_GTrsf

    - symbol: TopoDS
    - symbol: TopAbs_ShapeEnum
    - symbol: TopTools_ListOfShape
    - symbol: TopAbs_Orientation
    - symbol: TopLoc_Location
    - symbol: TopExp_Explorer

    - symbol: Poly_Connect
    - symbol: StdPrs_ToolTriangulatedShape

    - symbol: BRep_Tool
    - symbol: BRepTools
    - symbol: BRepGProp
    - symbol: BRepGProp_Face
    - symbol: GProp_GProps
    - symbol: BRepMesh_IncrementalMesh

    - symbol: BRepAdaptor_Curve
    - symbol: BRepAdaptor_CompCurve
    - symbol: GeomAbs_CurveType

    - symbol: StlAPI_Writer

    - symbol: BRepAlgoAPI_Cut
    - symbol: BRepAlgoAPI_Fuse

    - symbol: BRepBuilderAPI_Transform
    - symbol: BRepCheck_Analyzer
    - symbol: BRepPrimAPI_MakePrism
    - symbol: BRepOffsetAPI_MakeThickSolid
    - symbol: BRepOffsetAPI_MakeOffset
    - symbol: BRepOffset_Mode
    - symbol: BRepFilletAPI_MakeChamfer
    - symbol: ChFi3d_FilletShape
    - symbol: GeomAbs_JoinType
    - symbol: cadeau
  additionalCppCode: |
    #include <emscripten/val.h>
    class cadeau: public BRepTools {
    public:
      static emscripten::val UVBounds(const TopoDS_Face &f) {
        Standard_Real uMin;
        Standard_Real uMax;
        Standard_Real vMin;
        Standard_Real vMax;
        BRepTools::UVBounds(f, uMin, uMax, vMin, vMax);
        emscripten::val ret(emscripten::val::object());
        ret.set("uMin", emscripten::val(uMin));
        ret.set("uMax", emscripten::val(uMax));
        ret.set("vMin", emscripten::val(vMin));
        ret.set("vMax", emscripten::val(vMax));
        return ret;
      }

      static emscripten::val projectPointOnSurface( 
        const gp_Pnt & P,
        const opencascade::handle<Geom_Surface> & Surface
      ) {

        GeomAPI_ProjectPointOnSurf projectedPoint (P, Surface);

        Standard_Real u;
        Standard_Real v;

        projectedPoint.LowerDistanceParameters(u, v);

        emscripten::val ret(emscripten::val::object());
        ret.set("u", emscripten::val(u));
        ret.set("v", emscripten::val(v));
        return ret;
      }
    };
  emccFlags:
    - -sEXPORT_ES6=1
    - -sUSE_ES6_IMPORT_META=0
    - -sEXPORTED_RUNTIME_METHODS=["FS"]
    - -O3
```

P.S. If you want to have a look at what I am building, you can find a [preview here](https://cadeau.pages.dev/)

donalffons, 2021-06-17:
Nice example! Just out of curiosity - are those "real" parts?

For tracking down the issue, can you also share the JS code, which causes the error? Because the following works (when using a non-custom build)
```js
    const vec = new this.oc.gp_Vec_1();
    const xyz = vec.XYZ();
    console.log(xyz);
```

sgenoud, 2021-06-17:
> Nice example! Just out of curiosity - are those "real" parts?

Yes, kinda - it is a reimplementation of these parts (for a baby gym) [image]

The code that you pasted actually breaks on my machine...

```.js
  const vec = new oc.gp_Vec_1();
  console.log("ok", vec.X());
  const xyz = vec.XYZ();
  console.log("mmm", xyz);
```

[image]

donalffons, 2021-06-17:
Hmm, if I use the following code, I don't experience any issues... (Using a build with your `yml` made on my machine)

```js
    const oc = await openCascade({
      locateFile: () => "/cadeau_single.wasm",
    });
    console.log("START");
    const vec = new oc.gp_Vec_1();
    const xyz = vec.XYZ();
    console.log(xyz);
    console.log("END");
 ```
 
[image]

donalffons, 2021-06-17:
...however, using the files from you "Archive.zip", I can reproduce the error.

Can you try to `docker pull donalffons/opencascade.js` and then `docker run -it         --rm         -v $(pwd):/src         -u $(id -u):$(id -g)         donalffons/opencascade.js         buildConfig.yml` again?

sgenoud, 2021-06-17:
Just did a quick rebuild before going to bed - the error is still there unfortunately. did not seem to help...

```
 # docker pull donalffons/opencascade.js
Using default tag: latest
latest: Pulling from donalffons/opencascade.js
Digest: sha256:7af9122bcbb26a28eafe57f4fa0be92ff51101d3590e7ecd8e12a330759a0af4
Status: Image is up to date for donalffons/opencascade.js:latest
docker.io/donalffons/opencascade.js:latest
[Thu Jun 17 • 23:50:54] [~/workbench/cadeau] [deploy/baby-gym⚑]
 # cd wasm-build/
[Thu Jun 17 • 23:51:32] [~/workbench/cadeau/wasm-build] [deploy/baby-gym⚑]
 # docker run -it --rm -v $(pwd):/src -u $(id -u):$(id -g) donalffons/opencascade.js custom_build_single.yml
```

donalffons, 2021-06-18:
That's weird. I'm using the exact same image (same sha256), but I get (very slightly) different build results: [Archive.zip](https://github.com/donalffons/opencascade.js/files/6675217/Archive.zip)

Could you maybe upload the exact `yml` file that you are using?

What's your docker version?
Mine:
```
# docker --version
Docker version 20.10.7, build f0df35096d
```

Which OS are you using?
Mine:
```
# cat /etc/lsb-release
DISTRIB_ID=ManjaroLinux
DISTRIB_RELEASE=21.0.7
DISTRIB_CODENAME=Ornara
DISTRIB_DESCRIPTION="Manjaro Linux"
```

sgenoud, 2021-06-18:
```
 # docker --version
Docker version 20.10.6, build 370c289

 # sw_vers
ProductName:	Mac OS X
ProductVersion:	10.15.5
BuildVersion:	19F101
```
 updating my docker to match yours!

donalffons, 2021-06-18:
Interesting... I managed to reproduce your problem with a build made on an AWS EC2 mac1.metal instance:
```
# sw_vers 
ProductName:    macOS
ProductVersion: 11.4
BuildVersion:   20F71
```

I have no clue why that image would produce different results on MacOS. I will look into that further...

donalffons, 2021-06-18:
Argh, nevermind... I should have looked more carefully at the latest `yml` file that you posted.

Basically, that error message tells you that `gp_XYZ` is needed in that build. So just include it in your `yml` file, i.e.

```yml
 # cat custom_build_single.yml
cadeau_single.js:
  bindings:
    - symbol: Message_ProgressRange
    - symbol: TColgp_Array1OfDir

    - symbol: gp_XYZ # This was missing
    - symbol: gp_Vec
    - symbol: gp_Pnt
    - symbol: gp_Dir
    - symbol: gp_Ax2
    - symbol: gp_Ax3
    - symbol: gp_GTrsf

    - symbol: TopoDS
    - symbol: TopAbs_ShapeEnum
    - symbol: TopTools_ListOfShape
    - symbol: TopAbs_Orientation
    - symbol: TopLoc_Location
    - symbol: TopExp_Explorer

    - symbol: Poly_Connect
    - symbol: StdPrs_ToolTriangulatedShape

    - symbol: BRep_Tool
    - symbol: BRepTools
    - symbol: BRepGProp
    - symbol: BRepGProp_Face
    - symbol: GProp_GProps
    - symbol: BRepMesh_IncrementalMesh

    - symbol: BRepAdaptor_Curve
    - symbol: BRepAdaptor_CompCurve
    - symbol: GeomAbs_CurveType

    - symbol: StlAPI_Writer

    - symbol: BRepAlgoAPI_Cut
    - symbol: BRepAlgoAPI_Fuse

    - symbol: BRepBuilderAPI_Transform
    - symbol: BRepCheck_Analyzer
    - symbol: BRepPrimAPI_MakePrism
    - symbol: BRepOffsetAPI_MakeThickSolid
    - symbol: BRepOffsetAPI_MakeOffset
    - symbol: BRepOffset_Mode
    - symbol: BRepFilletAPI_MakeChamfer
    - symbol: ChFi3d_FilletShape
    - symbol: GeomAbs_JoinType
    - symbol: cadeau
  additionalCppCode: |
    #include <emscripten/val.h>
    class cadeau: public BRepTools {
    public:
      static emscripten::val UVBounds(const TopoDS_Face &f) {
        Standard_Real uMin;
        Standard_Real uMax;
        Standard_Real vMin;
        Standard_Real vMax;
        BRepTools::UVBounds(f, uMin, uMax, vMin, vMax);
        emscripten::val ret(emscripten::val::object());
        ret.set("uMin", emscripten::val(uMin));
        ret.set("uMax", emscripten::val(uMax));
        ret.set("vMin", emscripten::val(vMin));
        ret.set("vMax", emscripten::val(vMax));
        return ret;
      }

      static emscripten::val projectPointOnSurface(
        const gp_Pnt & P,
        const opencascade::handle<Geom_Surface> & Surface
      ) {

        GeomAPI_ProjectPointOnSurf projectedPoint (P, Surface);

        Standard_Real u;
        Standard_Real v;

        projectedPoint.LowerDistanceParameters(u, v);

        emscripten::val ret(emscripten::val::object());
        ret.set("u", emscripten::val(u));
        ret.set("v", emscripten::val(v));
        return ret;
      }
    };
  emccFlags:
    - -sEXPORT_ES6=1
    - -sUSE_ES6_IMPORT_META=0
    - -sEXPORTED_RUNTIME_METHODS=["FS"]
    - -O3
```

And then (hopefully), everything should be working fine :slightly_smiling_face:.

sgenoud, 2021-06-18:
🤦 

I made the change, but moved the file in the mean time and it saved the changes in the wrong place. This is why the first copy had the change, but not the file I was actually using...

This was too weird to be a proper bug... sorry to have made you loose your time (and thanks for your help)!

closed 2021-06-18 by sgenoud

donalffons, 2021-06-18:
No worries! Thanks for testing the beta version :wink:.