donalffons/opencascade.js #61: How to reduce the size of the wasm file? Improve the loading speed of the cad kernel? What is the best practice?

WREQI, 2021-06-23:
How to reduce the size of the wasm file?
Improve the loading speed of the cad kernel?
What is the best practice?

donalffons, 2021-06-23:
In that case you probably want to look into creating a [custom build](https://github.com/donalffons/opencascade.js/tree/master/doc#creating-a-custom-build). Just add whatever bindings you need to that yaml file and run the build as described. For smallest build size and fastest binaries, I would recommend to not use any of the MAIN_MODULE and SIDE_MODULE functionality that Emscripten provides.

The result will be just a "normal" Emscripten-compiled WASM binary with some JS glue code. You can import and use it [like this](https://github.com/emscripten-core/emscripten/issues/13074) in a Create-React-App.

I'm currently working on a re-write of that system for generating custom builds (mainly to speed it up). And I will probably make some (slight) changes to that YML file format for that same reason. Will update the docs when I make those changes.

WREQI, 2021-06-23:
I'm very happy for your answer, it doesn't sound so easy, I look forward to your opencascade.js update

closed 2021-06-24 by WREQI