donalffons/opencascade.js #62: Use Create Polygon Report Error

WREQI, 2021-06-24:
[image]
[image]

This is my code，Can someone tell me why the error was reported?

function Polygon(oc, points, wire) {
    let gpPoints = [];
    let curPolygon;
    for (var i = 0; i <points.length; i++){
    gpPoints.push(new oc.gp_Pnt_3(points[i].x,points[i].y,points[i].z))
    }
      let polygonWire = new oc.BRepBuilderAPI_MakeWire_1();
      for (let ind = 0; ind < points.length - 1; ind++) {
        let seg = new oc.GC_MakeSegment_1(gpPoints[ind], gpPoints[ind + 1]).Value();
        let edge = new oc.BRepBuilderAPI_MakeEdge_24(new oc.Handle_Geom_Curve_2(seg.get())).Edge();
        let innerWire = new oc.BRepBuilderAPI_MakeWire_2(edge).Wire();
        polygonWire.Add_2(innerWire);
      }
      let seg2 = new oc.GC_MakeSegment_1(gpPoints[points.length - 1], gpPoints[0]).Value();
      let edge2 = new oc.BRepBuilderAPI_MakeEdge_24(new oc.Handle_Geom_Curve_2(seg2.get())).Edge();
      let innerWire2 = new oc.BRepBuilderAPI_MakeWire_2(edge2).Wire();
      polygonWire.Add_2(innerWire2);
      let finalWire = polygonWire.Wire();
      if (wire) {
        curPolygon = finalWire;
      } else {
        curPolygon =new oc.BRepBuilderAPI_MakeFace_15(finalWire,true).Face();
      }
    return curPolygon;
  }

donalffons, 2021-06-24:
The constructors of `GC_MakeSegment` seems to throw an error if both points are identical. I.e.

```js
new this.oc.GC_MakeSegment_1(new this.oc.gp_Pnt_3(1, 2, 3), new this.oc.gp_Pnt_3(1, 2, 3)); // Throws error
new this.oc.GC_MakeSegment_1(new this.oc.gp_Pnt_3(1, 2, 3), new this.oc.gp_Pnt_3(2, 3, 4)); // Works fine
```

Maybe you should check that all your points have a sufficiently large distance between them.

It should also be possible to wrap some of your lines inside a `try { ... } catch() { ... }` block if you want to be able to react to errors in general.

WREQI, 2021-06-24:
Thank you very much, successfully solved my problem

closed 2021-06-25 by WREQI