donalffons/opencascade.js #70: Verify bindings breaks custom cpp code

sgenoud, 2021-07-05:
Using the latest docker build on the `test.yml` provided in the docs returns this

```
Traceback (most recent call last):
  File "/opencascade.js/src/buildFromYaml.py", line 66, in <module>
    verifyBindings(buildConfig["mainBuild"]["bindings"])
  File "/opencascade.js/src/buildFromYaml.py", line 64, in verifyBindings
    raise Exception("Requested binding " + json.dumps(binding) + " does not exist!")
Exception: Requested binding {"symbol": "CustomClass"} does not exist!
```

I guess you do no include the symbols in the additional cpp section in the verifyBindings section!

donalffons, 2021-07-05:
I see, thanks for reporting that. I made several changes to the build system recently and didn't test it well enough with `additionalCppCode`. There seem to be some other issues now, in addition to the one that you reported. Will let you know when this is resolved.

donalffons, 2021-07-07:
OK, this should now be fixed in the latest `@beta` build. [Here](https://github.com/donalffons/opencascade.js/commit/3ff2f9381de972565773eff9db79e47b9da2fd70) is the relevant commit, if you're interested to know. Basically, the logic I had in there previously was just plainly wrong.

The new logic is this:
1. If you're making a custom build, we inject your `additionalCppCode` into the `myMain.h` file, which is the entry point for clang to parse all the code. This file contains `#include' statements to all OCCT headers and (in case of a custom build) `additionalCppCode`.
2. If you're making a custom build, we only do binding generation and compilation for entities that are inside the `myMain.h` (since everything else has already been processed). Entities from other sources are skipped.
3. In the final final build step, the build system tries to find all entities mentioned in `bindings` in the YAML file and passes the relevant files to Emscripten. If it doesn't find anything, it raises the error you reported.

...Yet another reminder that this project does needs tests. I will open up some additional tickets to add tests.

Could you please try if this works now for you?

sgenoud, 2021-07-08:
Nice, thanks, it works!

closed 2021-07-08 by sgenoud