donalffons/opencascade.js #78: Open *.igs file got `getpwuid: TODO` error

J-F-Liu, 2021-07-28:
[image]

donalffons, 2021-07-28:
Thanks for reporting. That function `getpwuid` is currently not implemented in Emscripten. I have made a patch for this issue a while back [in this commit](https://github.com/donalffons/opencascade.js/commit/9416ee8f67e1b5249748f202a92f39d5c8117995), but for some reason it doesn't seem as if this is getting applied correctly.

I will look into that issue.

donalffons, 2021-09-07:
I added the patch back in - loading IGES files works now.
I will also add a test (tracking that [here](https://github.com/donalffons/opencascade.js/issues/74)) to make sure that "features" like this don't get removed accidentally, again.

closed 2021-09-07 by donalffons