donalffons/opencascade.js #85: Would like instructions for how to incorporate into create-react-app

MattFerraro, 2021-09-02:
Many developers use create-react-app to get a new project started. It would be great to have a section in the readme describing exactly what they need to do in order to get opencascade working inside a new CRA project.

I'm currently trying to figure it out. Will be happy to make a PR documenting it if I can manage to make it work!

The main difficulty is that CRA doesn't give you direct access to the webpack file, so I can't figure out how to easily add wasm support for bundling.

MattFerraro, 2021-09-02:
I've put together a tiny repo [here](https://github.com/MattFerraro/openCascadeTest1) to try and get this working.

I've installed react-app-rewired which lets me modify the webpack config, then added the appropriate lines in config-overrides.js. From there I installed opencascade.js and try to import it but in the console I see: 
[image]

Not sure where to go from here. Any advice appreciated!

donalffons, 2021-09-02:
Hi! Having some more detailed documentation on how to use this library with React (and / or a link to an example repository) will be a great addition.

These error messages mean, that the Webpack-generated URL to the `.wasm` file points nowhere. In this case, the browser will try to request `localhost:8000/nowhere.wasm` and because this is a single page app with client side routing, the Webpack dev server responds with the "index" html document. The browser than complains about this not being a valid `.wasm` file.

Here is a `config-overrides.js` (used by react-app-rewired) that works for me.
```js
module.exports = {
  webpack: (config) => {
    config.module.rules.find(k => k.oneOf !== undefined).oneOf.unshift(
      {
        test: /\.wasm$/,
        type: "javascript/auto",
        loader: "file-loader",
        options: {
          name: "static/js/[name].[contenthash:8].[ext]",
        },
      }
    );
    return config;
  },
};
```

Create-React-App currently uses Webpack 4. Starting from Webpack 5, there is support for the `import.meta` syntax, which (when also enabled in Emscripten)  should remove the need for any "manual" configuration. There is lots of discussion online, indicating the Create-React-App is working on transitioning to Webpack 5.

MattFerraro, 2021-09-03:
That worked like a charm for me! Thanks for your help

closed 2021-09-03 by donalffons