donalffons/opencascade.js #92: How to use methods that take a stream (ie DumpJson)

trungleduc, 2021-11-27:
Hello, thank you for a great library. 
I'm new to WebAssembly and i want to dump a shape to json file. But this method takes a Standard_OStream object as input and i don't know how to create it in js.
Do you have any hint to do it? Thank you very munch.

donalffons, 2021-11-27:
Hi, that is something that is currently not implemented, i.e. you cannot use these functions directly from JavaScript.

However, you can create a [custom build](https://github.com/donalffons/opencascade.js/tree/master/doc#creating-a-custom-build) of OpenCascade.js and implement your own abstractions around any DumpJson method in C++ using the `additionalCppCode` field. The bindings between C++ and JS will then be generated automatically using [Embind](https://emscripten.org/docs/porting/connecting_cpp_and_javascript/embind.html). In case you want to return a `std::string` from C++ to JS, that should work out of the box.

If you want to create a "full" custom build, similar to the builds published on npm, have a look at the comments in [this issue](https://github.com/donalffons/opencascade.js/issues/91#issuecomment-949534411). You will need a development machine with a lot of memory in this case.

donalffons, 2021-11-27:
As a side note: Are you aware of OpenCascade's own `xbf` file format? I posted a short snippet that shows how to write `xbf` files (and how to write and read `glb` files) [here](https://github.com/donalffons/opencascade.js/discussions/89#discussioncomment-1374536). Depending on what you want to do, this might be a better approach?! From what I saw online, the DumpJson methods are not well documented and [might not be stable](https://dev.opencascade.org/content/how-use-dumpjson).

trungleduc, 2021-11-29:
Thank you very much for your highlight, indeed `xbf` format is what I need!

closed 2021-11-29 by trungleduc