donalffons/opencascade.js #94: Error while launching Docker container

womaas, 2021-12-25:
HI. When running donalffons/opencascade.js using Docker Desktop, the following error occurs:

```
File "/opencascade.js/src/buildFromYaml.py", line 16, in <module>
buildConfig = yaml.safe_load(open(sys.argv[1], "r"))
IndexError: list index out of range
```

How can this be fixed?

donalffons, 2021-12-26:
Hi, did you follow the instructions in the [docs](https://github.com/donalffons/opencascade.js/tree/master/doc#creating-a-custom-build)? You have to supply the file name of your custom build YAML file (which should be in the current directory, from where you run the build). The error message looks like you didn't supply a file name.

womaas, 2021-12-26:
Thanks. I am trying to start Docker this way, but I get an error: 
`unknown shorthand flag: 'g' in -g)`

donalffons, 2021-12-26:
Hmm, maybe your linux distribution doesn't know the command `id -g`?
Try the command without that line, i.e.
```
docker run \
  --rm \
  -it \
  -v $(pwd):/src \
  donalffons/opencascade.js \
  test.yml
```

I believe this will create the build files with root ownership. You might want to change the ownership to your user via `sudo chown user:user filename` after the build is finished.

womaas, 2021-12-26:
```
docker: Error response from daemon: create $(pwd): "$(pwd)" includes invalid characters for a local volume name, only "[a-zA-Z0-9][a-zA-Z0-9_.-]" are allowed. If you intended to pass a host directory, use absolute path.
See 'docker run --help'.
```

donalffons, 2021-12-26:
Which OS are you using?
It seems like your OS doesn't know how to interpret `$(pwd)`. In that case, you can replace `$(pwd)` with the full path to the directory in which the YAML file is located, i.e. `/home/users/yourusername/myproject/`.

womaas, 2021-12-26:
Windows.
The full path works. But: 
```
...
File "/usr/local/lib/python3.8/dist-packages/yaml/parser.py", line 438, in parse_block_mapping_key
    raise ParserError("while parsing a block mapping", self.marks[-1],
yaml.parser.ParserError: while parsing a block mapping
  in "test.yml", line 1, column 1
expected <block end>, but found '<block mapping start>'
  in "test.yml", line 39, column 3
```

**line 39** is
`additionalCppCode: |`

donalffons, 2021-12-26:
[Here](https://stackoverflow.com/questions/38272137/yaml-parser-parsererror-while-parsing-a-block-mapping) is a stackoverflow issue that suggests that this might be related to using tabs (instead of spaces) in your YAML file.

womaas, 2021-12-26:
Not tabs, but spaces (file copied from [here](https://github.com/donalffons/opencascade.js/tree/master/doc#creating-a-custom-build)). Everything works well with no spaces on line 39.

donalffons, 2022-01-07:
I think in, the following should just work (but I didn't test it).
```yml
    bindings:
      - symbol: Handle_Geom2d_Curve
      // ...
```
This should work, because of [this line](https://github.dev/Open-Cascade-SAS/OCCT/blob/16a263dc17600ac13a5110bd7028ba558d1a0700/src/Geom2d/Geom2d_Curve.hxx#L34) in the OCCT headers, which is a macro that #defines the `Handle_` type.

womaas, 2022-01-09:
Thanks, I'll check. The question is off topic.
Some classes have methods that are called like this:
`aFixWire-> DropSmallEdgesMode () = true;`

DropSmallEdgesMode description looks like:
```
FixFreeWireMode ()
Standard_Integer & ShapeFix_Shape :: FixFreeWireMode ()

Returns (modifiable) the mode for applying fixes of ShapeFix_Wire, by default True.
```

In opencascade.full.d.ts:
```
export declare class ShapeFix_Shape extends ShapeFix_Root {
...
FixFreeWireMode (): Graphic3d_ZLayerId;
```

How to set values in JS for such cases?

donalffons, 2022-03-29:
Sorry for the late reply.

This is currently not "natively" supported by OpenCascade.js. As a work-around, you can create a custom build and make the un-supported calls from there.

I added your suggestion to the To-Do list. This is something that might be supported in the future.

closed 2022-03-29 by donalffons