donalffons/opencascade.js #98: IntTools_Context no constructors available

bitbybit-dev, 2022-01-31:
Hey, I've been trying to implement point in face feature [BOPTools_AlgoTools3D.PointInFace_2](https://dev.opencascade.org/doc/refman/html/class_b_o_p_tools___algo_tools3_d.html#a4b3a5ace07466fcefe4cc711e353573b) and it needs IntTools_Context handle, but while Handle_IntTools_Context constructors exist on occ, there is no IntTools_Context constructor to make the new context. 

If I understand correctly those contexts are  used mostly for caching, maybe there's some global context that I could tap into to construct the handle? Let me know if I should ask this on OCCT forum... Thanks.

bitbybit-dev, 2022-02-02:
ah, nevermind, found that it's excluded in filterClasses

  # error: call to implicitly-deleted copy constructor of 'IntTools_FClass2d'
  # error: call to implicitly-deleted copy constructor of 'BRepClass3d_SolidClassifier'
  if theClass.spelling == "IntTools_Context":
    return False

donalffons, 2022-03-29:
Sorry for the late reply! Thanks a lot for these bug reports / feature requests.

I just made the following quick fix: Instead of filtering out the entire class, I now only filter out the problematic methods.

Will post again, when the build has gone through.

donalffons, 2022-03-30:
The latest build `2.0.0-beta.a869126`has the changes. Please let me know if that fixes your problem. Cheers!

donalffons, 2022-03-30:
Also: I recently upgraded to the latest version of OpenCascade, which introduced several breaking changes, due to which I had to update the tests and examples. You will probably experience similar issues if you haven't upgraded the package within the last couple of weeks or so.

bitbybit-dev, 2022-03-30:
Thanks for the heads up. Will prepare to dedicate some time for migration ;) Good to know you're on the latest version.

donalffons, 2022-05-05:
The following code works fine in the current version:
```ts
const handle = new oc.Handle_IntTools_Context_2(new oc.IntTools_Context_1());
```
I haven't tried to do any meaningful with it yet, but I believe this issue can be closed. Feel free to reopen, if you're having trouble.

closed 2022-05-05 by donalffons