donalffons/opencascade.js #103: --shared-memory is disallowed because it was not compiled with 'atomics' or 'bulk-memory' features.

aihornmac, 2022-03-30:
Thx so much for your work on this lib! It helped a lot!

Now I wanna go further with multi threading, so I tried to build from docker. Unfortunately it failed.

I was hoping that you could shed some light on it.

Here is the error message:

```
wasm-ld: error: --shared-memory is disallowed by /opencascade.js/build/bindings/BRepTools/BRepTools.hxx/BRepTools.cpp.o because it was not compiled with 'atomics' or 'bulk-memory' features.
emcc: error: '/emsdk/upstream/bin/wasm-ld @/tmp/emscripten_8un2px8q.rsp.utf-8' failed (returned 1)
```
Full file see: [error.log](https://github.com/donalffons/opencascade.js/files/8383473/error.log)

Here is the command I ran:

```bash
docker run --rm -e threading=multi-threaded -it -v $(pwd):/src -u $(id -u):$(id -g) donalffons/opencascade.js build/occ.yaml
```

with following yaml file:

```yaml
  mainBuild:
    name: rocketExample.js
    bindings:
      # - symbol: STEPCAFControl_Reader
      # - symbol: IFSelect_ReturnStatus
      # - symbol: TCollection_ExtendedString
      # - symbol: TDocStd_Document
      # - symbol: Handle_TDocStd_Document
      - symbol: Message_ProgressRange
      # - symbol: XCAFDoc_DocumentTool
      # - symbol: TDF_LabelSequence
      # - symbol: XCAFDoc_ShapeTool
      # - symbol: TCollection_AsciiString
      # - symbol: RWGltf_CafWriter
      - symbol: BRepTools
      - symbol: BRepMesh_IncrementalMesh
      # - symbol: TColStd_IndexedDataMapOfStringString
      # - symbol: CDM_Document
      # - symbol: Standard_Transient
      # - symbol: TDF_Label
      # - symbol: TDataStd_GenericEmpty
      # - symbol: TDF_Attribute
      # - symbol: Handle_XCAFDoc_ShapeTool
      # - symbol: NCollection_BaseSequence
      - symbol: TopoDS_Shape
      # - symbol: BRepMesh_DiscretRoot
      # - symbol: NCollection_BaseMap

      - symbol: CustomClass
    emccFlags:
      # @see https://github.com/emscripten-core/emscripten/blob/main/src/settings.js
      - -O3
      - -sEXPORT_ES6=1
      - -sUSE_ES6_IMPORT_META=0
      - -sEXPORTED_RUNTIME_METHODS=['FS']
      - -sINITIAL_MEMORY=100MB
      - -sMAXIMUM_MEMORY=4GB
      - -sALLOW_MEMORY_GROWTH=1
      - -sUSE_FREETYPE=1
      # - -sWASM_WORKERS=1
      # - -pthread
      # - -sPTHREAD_POOL_SIZE='navigator.hardwareConcurrency'
      # - -sOFFSCREENCANVAS_SUPPORT=1
  additionalCppCode: |
    #include <iostream>
    class CustomClass {
    public:
      static void SayHello() {
        std::cout << "Hello, World" << std::endl;
      }
    };
```

aihornmac, 2022-03-31:
@donalffons

I browsed thru the source code and found the way to make it work. Building with a subset of bindings did complete, but when I tried to build with full bindings, it got killed with -9

full log: [error.log](https://github.com/donalffons/opencascade.js/files/8391294/error.log)

Here is the command I ran:
```bash
docker run -v $(pwd):/src -u $(id -u):$(id -g) donalffons/opencascade.js:multi-threaded build/opencascade.full.yml
```

`opencascade.full.yml` is a copy from opencascade.js source code

I wonder what am I missing.

donalffons, 2022-03-31:
Hey,
thanks for testing this feature. The multi-threaded builds were broken and unusable until a few hours ago. They should now be working.

I also just updated [the docs on multi-threading](https://github.com/donalffons/opencascade.js/tree/master/doc#multi-threading), which were missing some important bits of information. And there is now a [passing test](https://github.com/donalffons/opencascade.js/blob/master/test/multi-threaded.test.ts) that you can take as a reference.

> but when I tried to build with full bindings, it got killed with -9

You're probably running out of memory during the build?! Making a full build currently takes a huge amount of memory (~40 Gigs on my system). If you don't have this amount of memory available on your system, you might be able to use swap space on Linux and MacOS or virtual memory on Windows.

Eventually, I want to distribute a full multi-threading-enabled version of OpenCascade.js via npm. But right now, there are a few other issues I want to focus on.

aihornmac, 2022-03-31:
您的邮件已被接收，感谢您的来信。您的郵件已被接收，感謝您的來信。Your mail has been received, thanks for your letter.あなたのメールは受信されました、あなたの手紙をありがとう。?3?3?0?3?3?1?3?3 ?2?1?2?1?9?7?2?5?2?8?9?1?9?9. ?9?0?2?1?3?5 ?5?3?3?1?2?3 ?7?1?1?7?9?3?0?2?9?1?9?9.Votre e-mail a été re?0?4u. Merci pour votre lettre.Ваше сообщение было получено. Спасибо за ваше письмо.

aihornmac, 2022-03-31:
@donalffons 
I raised up the resources of Docker as you mentioned, and it succeeded! Thanks so much for your kind help!