donalffons/opencascade.js #106: Export to STEP with STEPCAFControl_Writer

JonOsterman19, 2022-04-07:
I'm currently in the process of exporting to STEP but I have a segmentation fault error with STEPCAFControl_Writer. Below you can find detailed steps for what I'm doing so far. I'm using the latest docker image to compile, below more details of what I'm using

- Using Open Cascade version 7.6.1 │ Commit d2abb6d844231cb8f29be6894440874a4700e4a5
- https://hub.docker.com/layers/opencascade.js/donalffons/opencascade.js/latest/images/sha256-f92022555539f04251de09fcf6c91f29d346c439eeab5b1a6212e7279a609a20?context=explore

Made a simple cube in Blender and exported as Gltf (Below an image of the settings) [https://siasky.net/OADjO_xqTABc1CTM_z_W59_QQEft7yrWaBXOuekTcAaTUA]

```
// Setup XCAF document
const app = new openCascade.TDocStd_Application();
const doc = new openCascade.Handle_TDocStd_Document_1();
const format = new openCascade.TCollection_ExtendedString_2("MDTV-XCAF", true);
app.NewDocument_2(format, doc);

// Create data file
var fileString = this.file // imported gltf file
openCascade.FS.createDataFile("/", "file.gltf", fileString, true, true);

// Read Gltf file
var readerGltf = new openCascade.RWGltf_CafReader()
readerGltf.SetSkipEmptyNodes(false)
readerGltf.SetMeshNameAsFallback(true)
readerGltf.Metadata()
readerGltf.SetDocument(doc)

// Translation to OCCT model
const str = new openCascade.TCollection_ExtendedString_2("/file.gltf", true);
const fileStr = new openCascade.TCollection_AsciiString_13(str, 0);
const progress = new openCascade.Message_ProgressRange_1()

readerGltf.Perform(fileStr, progress)
```
So far so good. I can get all the shapes with GetFreeShapes and loop through them to get naming and assembly structure. So I know the document has the shapes.

The problem I have is with **STEPCAFControl_Writer.Transfer_1** and **STEPCAFControl_Writer.Perform_2**. I tested both and first I got a null error. I added **-sASSERTION=1**, **SAFE_HEAP=1** and **-sNO_DISABLE_EXCEPTION_CATCHING** to the compiler settings and then I got the segmentation fault error.

I tested **Transfer_1** with **STEPControl_AsIs** and **Perform_2**. These 2 give the same "segmentation fault" error.
With **Transfer_1** and **STEPControl_ManifoldSolidBrep** it exports the STEP file with the right assembly structure but the STEP is empty because I pass a string to multi and therefore uses external STEP files that are not exported with the root step file. I also have no idea how I can access them to check if they where exported correctly.

This is from the docs

> Transfers a document (or single label) to a STEP model The mode of translation of shape is AsIs If multi is not null pointer, it switches to multifile mode (with external refs), and string pointed by <multi> gives prefix for names of extern files (can be empty string) Returns True if translation is OK.

Below the code to export to STEP

```
// Export to STEP
var fileLocation = "/model.step"
var writerCAF = new openCascade.STEPCAFControl_Writer_1()
writerCAF.SetColorMode(true);
writerCAF.SetNameMode(true);

openCascade.Interface_Static.SetIVal("write.precision.mode", 1);
openCascade.Interface_Static.SetIVal("write.precision.val", 0.1);
openCascade.Interface_Static.SetIVal("write.step.assembly", 1);
openCascade.Interface_Static.SetIVal("write.step.unit", this.units[unit].index);
openCascade.Interface_Static.SetIVal("write.step.schema", 4);
openCascade.Interface_Static.SetIVal("write.step.vertex.mode", 0);
openCascade.Interface_Static.SetIVal("write.surfacecurve.mode", 1);
openCascade.Interface_Static.SetIVal("write.stepcaf.subshapes.name", 1);

var progressOutput = new openCascade.Message_ProgressRange_1()

//var written = writerCAF.Transfer_1(doc, openCascade.STEPControl_StepModelType.STEPControl_AsIs, "", progressOutput);
//var written = writerCAF.Transfer_1(doc, openCascade.STEPControl_StepModelType.STEPControl_ManifoldSolidBrep, "", progressOutput);
var written = writerCAF.Perform_2(doc, fileLocation, progressOutput)
```

donalffons, 2022-04-07:
> I tested both and first I got a null error. I added -sASSERTION=1, SAFE_HEAP=1 and -sNO_DISABLE_EXCEPTION_CATCHING to the compiler settings and then I got the segmentation fault error.

Maybe that's a typo, but the [assertion setting](https://emsettings.surma.technology/#ASSERTIONS) is spelled `-sASSERTIONS=1`. Setting `sNO_DISABLE_EXCEPTION_CATCHING` is likely to cause problems as this flag would have to be provided both at compile and link time. If you're creating a custom build, this setting would only get applied at link time - so the linker might expect exception catching code to be available - and when you're encountering exceptions during run-time, it fails. But that's just a guess.

One reason why you're currently not getting good error messages is that OpenCascade.js comes without support for exception catching. The reason that feature is disabled is because Emscripten's non-native exceptions greatly affect both file size and runtime performance. The ideal solution would be to use native exceptions, but that's [not yet supported in Firefox](https://webassembly.org/roadmap/).

But I realize that having no exceptions available at all might be a bit of a deal breaker. I will make a few changes and get back to you in a bit.

> With Transfer_1 and STEPControl_ManifoldSolidBrep it exports the STEP file with the right assembly structure but the STEP is empty because I pass a string to multi and therefore uses external STEP [...]

You could try using `oc.FS.readdir("/")` which will give you a string array with the directory contents, so you could at least check if these additional files get exported.

donalffons, 2022-04-07:
(I just opened an [issue over at Emscripten](https://github.com/emscripten-core/emscripten/issues/16665) with an idea for a more long-term solution)

Meanwhile, there's a build running with support for non-native exceptions.

JonOsterman19, 2022-04-07:
> 

Thank you for the quick reply. I tried `oc.FS.readdir("/")` and got `['.', '..', 'tmp', 'home', 'dev', 'proc']` so I guess the external .stp files are not saved. That is also probably the reason why there is no error when exporting to STEP

donalffons, 2022-04-08:
Alright, after some research into this, I don't think you would be getting improved error message for this particular case. However, I did push a build to dockerhub with the tag [`donalffons/opencascade.js:test-non-native-exceptions`](https://hub.docker.com/layers/201818244/donalffons/opencascade.js/test-non-native-exceptions/images/sha256-8fafcc87683a0f8d57ee9cf66de014d274da9c2c48ac302756d9b08ca3d8dc6b?context=repo). That has been compiled with the flag `-fexceptions` which enables [non-native exceptions](https://emscripten.org/docs/porting/exceptions.html#javascript-based-exception-support). Unfortunately, the "full" builds produced by this Image don't run in V8 (Chrome, Node), because the maximum function count is exceeded. I will work on finding a better way of integrating exception handling into this project, but that might take a while.

Back to your error: I tracked this down and found that it is failing with a segfault [in this line in Standard_Transient.cxx](https://github.com/Open-Cascade-SAS/OCCT/blob/V7_6_1/src/Standard/Standard_Transient.cxx#L56), at the call to `DynamicType()`, which seems odd. I don't yet understand why that is happening and I will work on creating a reduced test case.

In case you're experienced with OpenCascade in C++, would you be able to re-write your simple test case with "native" OpenCascade to see how it behaves there and share the results here?

As a workaround: It would probably be possible to manually convert your GLTF model into a bunch of [`BRepBuilderAPI_MakePolygon`s](https://ocjs.org/docs/examples/polygon) - although that's certainly not a very elegant solution.

JonOsterman19, 2022-04-08:
> 

The new docker image with exceptions is very helpful for debugging, thank you!

I have never written C++ so I can't really help there.

It seems that when ever it sends data to the worksession it crashes.

Below the result of the simple shape constructed with OCC
[image]

And here is the result whenever I try to export to STEP with the gltf converted to the OCCT model
[image]

donalffons, 2022-04-08:
Thanks for sharing the actual error messages. These are the same I can reproduce in NodeJS.

> Also when I try the code below I get the same abort segmentation error on `writerCAF.Init(WS, true)`
> 
> ```
> var WS = new openCascade.Handle_XSControl_WorkSession_1()
> var writerCAF = new openCascade.STEPCAFControl_Writer_1()
> writerCAF.Init(WS, true)
> ```

This seems to be a different issue (the segfault doesn't happen in the same place). My guess is that you should at least use `var WS = new openCascade.Handle_XSControl_WorkSession_2(new openCascade.XSControl_WorkSession())` in your first line - otherwise you're passing a handle with no target object to the `Init` method.

donalffons, 2022-04-09:
After a bit more research, I found that the `this` pointer in the [problematic line in Standard_Transient.cxx](https://github.com/Open-Cascade-SAS/OCCT/blob/V7_6_1/src/Standard/Standard_Transient.cxx#L56) is actually null. I believe calling a method on a null pointer is undefined behavior in C++, so it's OK for this to segfault. Therefore, I don't believe that this is an error / problem with OpenCascade.js and rather just how things are implemented in OpenCascade / OCCT.

I also tried importing a GLTF file with [OpenCascade's CADAssistant](https://www.opencascade.com/products/cad-assistant/), which suggests that this use case isn't really supported: After importing a GLTF file, it doesn't allow you to export in STEP (or any kind of BREP format) any more.

There seems to be some attribute in the imported STEP data that tips off OpenCascade.

I would recommend to convert the mesh data in the GLTF into `TopoDS_Face`s (like in the polygon example, I linked to).

JonOsterman19, 2022-04-12:
> After a bit more research, I found that the `this` pointer in the [problematic line in [...]

Thank you for looking into this. I'm also thinking it is not really a use case that I will keep exploring. It makes sense that it is not really possible because of the fact that you can make a certain shape in alot of different ways and a piece of code can't really know what the best way is. That is my interpretation of it. Like you suggested reconstructing the faces is a possibility but your files just explode in size because for every face in your mesh it generates a lot of data.

I found this online that has a good explanation of what is going on. For small parts it can be done but that is not really what I was looking for.

https://transmagic.com/can-i-convert-polygonal-models-to-cad-models/

Again thank you for looking into this and keep up the great work!

closed 2022-04-12 by donalffons