donalffons/opencascade.js #117: Fix typescript definitions for enums

donalffons, 2022-05-04:
The following code
 ```ts
const quat = new oc.gp_Quaternion_1();
quat.SetEulerAngles(oc.gp_EulerSequence.gp_Extrinsic_XYZ, 0, 0, 0));
 ```
results in a typescript error:
```
(property) gp_Extrinsic_XYZ: {}
Argument of type '{}' is not assignable to parameter of type 'gp_EulerSequence'.
  Type '{}' is missing the following properties from type 'gp_EulerSequence': gp_EulerAngles, gp_YawPitchRoll, gp_Extrinsic_XYZ, gp_Extrinsic_XZY, and 22 more.ts(2345)
```

donalffons, 2022-05-22:
The previous syntax in the "valid code" was incorrect. Instead, this should be:
```js
const quat = new oc.gp_Quaternion_1();
quat.SetEulerAngles(oc.gp_EulerSequence.gp_Extrinsic_XYZ.value, 0, 0, 0));
```

It's tricky to figure this out, since Emscripten doesn't seem to error. Another example:

The following code causes no typescript error and no runtime error. It is however, incorrect.
```js
myContext.SetDisplayMode_1(oc.AIS_DisplayMode.AIS_Shaded, true);
console.log(myContext.DisplayMode()); // outputs '0', should be '1'
```

The following code causes a typescript error, but works correctly at runtime
```js
myContext.SetDisplayMode_1(oc.AIS_DisplayMode.AIS_Shaded.value, true);
console.log(myContext.DisplayMode()); // outputs '1'
```

ghdalsrl326, 2024-07-08:
I got a similar error when I tried to use oc.TopAbs_ShapeEnum.TopAbs_FACE. How can I fix it?

TS2345: Argument of type '{}' is not assignable to parameter of type 'TopAbs_ShapeEnum'.
  Type '{}' is missing the following properties from type 'TopAbs_ShapeEnum': TopAbs_COMPOUND, TopAbs_COMPSOLID, TopAbs_SOLID, TopAbs_SHELL, and 5 more.
  
ocjs  version:  "opencascade.js": "^2.0.0-beta.b5ff984",