donalffons/opencascade.js #120: Expected null or instance of Handle_HLRBRep_PolyAlgo, got an instance of Standard_Transient

atlas1119, 2022-05-05:
do not support HLR?

atlas1119, 2022-05-05:
my code:
const myPolyAlgo = new openCascade.HLRBRep_PolyAlgo_1();
  myPolyAlgo.Load(shape);

  const gp_Ax2 = new openCascade.gp_Ax2_3(new openCascade.gp_Pnt_3(0.0, 0.0, 0.0), new openCascade.gp_Dir_4(0.0, 1.0, 0.0))
  const myProjector = new openCascade.HLRAlgo_Projector_2(gp_Ax2);
  myPolyAlgo.Projector_2(myProjector);
  myPolyAlgo.Update();
  const aPolyHLRToShape = new openCascade.HLRBRep_PolyHLRToShape(); 
  aPolyHLRToShape.Update(myPolyAlgo);

donalffons, 2022-05-05:
Try wrapping it in `Handle_HLRBRep_PolyAlgo`, e.g. `aPolyHLRToShape.Update(new openCascade.Handle_HLRBRep_PolyAlgo_2(myPolyAlgo));`

atlas1119, 2022-05-06:
> 

thank you @donalffons