donalffons/opencascade.js #135: Add typescript definitions for extracting exception data

JohannesMerkt, 2022-05-24:
As described in https://ocjs.org/docs/advanced/exceptions/catch-exceptions you can use 
```js
try {
  const disc = new oc.BRepPrimAPI_MakeCone_1(1, 0, 0);
} catch (e) {
  if(typeof e === "number") {
    const exceptionData = oc.OCJS.getStandard_FailureData(e);
    console.log(`That didn't work because: ${exceptionData.GetMessageString()}`);
  } else {
    console.log("Unkown error");
  }
}
```
to retrive the actual exception data the error number points to. However the typescript definition does not know about OCJS and its sub functions. So workarounds are needed
```ts
try {
  const disc = new oc.BRepPrimAPI_MakeCone_1(1, 0, 0);
} catch (e) {
  if(typeof e === "number") {
    const exceptionData = (oc as any).OCJS.getStandard_FailureData(e);
    console.log(`That didn't work because: ${exceptionData.GetMessageString()}`);
  } else {
    console.log("Unkown error");
  }
}
```

JohannesMerkt, 2022-05-24:
If it is as simple as adding those to the opencascade.full.d.ts file in typedoc-reference-docs. I could create a merge request

donalffons, 2022-05-25:
This works now in `2.0.0-beta.ae74abe`.
[image]

commit 00798adb (2022-05-24): move OCJS helper code into `additionalCppCode`, fixes #135
closed 2022-05-25 by donalffons