donalffons/opencascade.js #154: Cannot call BRep_Tool.Triangulation due to unbound types: N11opencascade6handleI18Poly_TriangulationEE

flolu, 2022-07-14:
I'm using the following code from [openCascadeHelper.js](https://github.com/donalffons/opencascade.js-examples/blob/e675ed7ea9287356df23e2fda496cfe735f86f23/src/common/openCascadeHelper.js#L17)

```typescript
const myFace = oc.TopoDS.Face_1(ExpFace.Current())
const aLocation = new oc.TopLoc_Location_1()
const myT = oc.BRep_Tool.Triangulation(myFace, aLocation, 0 /* == Poly_MeshPurpose_NONE */)
```

But the third line throws the following error:

```
UnboundTypeError: Cannot call BRep_Tool.Triangulation due to unbound types: N11opencascade6handleI18Poly_TriangulationEE
```

**Which bindings am I missing?** Is there a list of all available bindings? Note: the code works fine with the full build.

This is my current custom build file:
```yaml
mainBuild:
  name: customBuild.examples.js
  bindings:
    - symbol: BRepTools
    - symbol: NCollection_BaseSequence
    - symbol: TDataStd_GenericEmpty
    - symbol: TDF_Attribute
    - symbol: TDF_LabelSequence
    - symbol: XSControl_Reader
    - symbol: STEPControl_Reader
    - symbol: IFSelect_ReturnStatus
    - symbol: RWGltf_CafWriter
    - symbol: TCollection_AsciiString
    - symbol: Handle_TDocStd_Document
    - symbol: TDocStd_Document
    - symbol: TColStd_IndexedDataMapOfStringString
    - symbol: Message_ProgressRange
    - symbol: XCAFDoc_DocumentTool
    - symbol: XCAFDoc_ShapeTool
    - symbol: BRepMesh_IncrementalMesh
    - symbol: TopoDS_Shape
    - symbol: Handle_XCAFDoc_ShapeTool
    - symbol: STEPCAFControl_Reader
    - symbol: NCollection_BaseMap
    - symbol: TDF_Label
    - symbol: BRepMesh_DiscretRoot
    - symbol: TCollection_ExtendedString
    - symbol: CDM_Document
    - symbol: Standard_Transient
    - symbol: TopExp_Explorer
    - symbol: TopAbs_ShapeEnum
    - symbol: TopoDS_Compound
    - symbol: TopoDS_Face
    - symbol: TopoDS_Builder
    - symbol: TopoDS_Edge
    - symbol: TopoDS_Wire
    - symbol: TopoDS
    - symbol: TopoDS_Iterator
    - symbol: TopLoc_Location
    - symbol: Poly_Connect
    - symbol: BRep_Tool
    - symbol: TColgp_Array1OfDir
    - symbol: StdPrs_ToolTriangulatedShape
    - symbol: Poly_Triangulation
    - symbol: RWMesh_TriangulationSource
  emccFlags:
    - -flto
    - -fexceptions
    - -sDISABLE_EXCEPTION_CATCHING=1
    - -O3
    - -sEXPORT_ES6=1
    - -sUSE_ES6_IMPORT_META=0
    - -sEXPORTED_RUNTIME_METHODS=['FS']
    - -sINITIAL_MEMORY=15MB
    - -sMAXIMUM_MEMORY=4GB
    - -sALLOW_MEMORY_GROWTH=1
    - -sLLD_REPORT_UNDEFINED
    - --no-entry
    - -sENVIRONMENT='web'
```

donalffons, 2022-07-14:
`N11opencascade6handleI18Poly_TriangulationEE` is the mangled type name. You can use a [Demangler](http://demangler.com/) to get the original type name by prepending `_Z`, i.e. `_ZN11opencascade6handleI18Poly_TriangulationEE`. This gives you `opencascade::handle<Poly_Triangulation>`. That type has already been explicitly defined [here in OCCT](https://github.com/Open-Cascade-SAS/OCCT/blob/V7_6_2/src/Poly/Poly_Triangulation.hxx#L34) with a macro that defines it as `Handle_Poly_Triangulation`. To make template types usable in JS, the template argument has to be resolved by using a typedef.

Long story short: I think you need to add `- symbol: Handle_Poly_Triangulation`.

If you ever need to use template types that haven't been defined in OCCT in a similar way, check the full build - it contains a typedef for `Handle_IMeshTools_Context`.

flolu, 2022-07-14:
Thank you so much for your quick response! This worked.

---

But what about `18NCollection_Array1I13Poly_TriangleE`?

The demangler outputs: `NCollection_Array1<Poly_Triangle>`, but the symbol `NCollection_Array1_Poly_Triangle` does not exist. `NCollection_Array_Poly_Triangle` does not exist, either.

I've also tried `BRepMesh_Triangulator` as defined [here in OCCT](https://github.com/Open-Cascade-SAS/OCCT/blob/5614b1369a232f43e59cb9be4f5f8efd2075608d/src/BRepMesh/BRepMesh_Triangulator.hxx#L32). But I still get the error: `UnboundTypeError: Cannot call Poly_Triangulation.Triangles due to unbound types: 18NCollection_Array1I13Poly_TriangleE`

donalffons, 2022-07-14:
> But what about 18NCollection_Array1I13Poly_TriangleE?

That is typedef'd [here](https://github.dev/Open-Cascade-SAS/OCCT/blob/33c8a72788661285860902f261062e39f32a2d46/src/Poly/Poly_Array1OfTriangle.hxx#L23) as `Poly_Array1OfTriangle`.

donalffons, 2022-07-14:
The type can also be seen in typescript definitions (`opencascade.d.ts` via npm) and the [reference docs](https://ocjs.org/reference-docs/classes/Poly_Triangulation#Triangles).

flolu, 2022-07-14:
Thank you so much!

closed 2022-07-14 by flolu