donalffons/opencascade.js #156: Read STEP Instance ID's from Shapes

flolu, 2022-07-18:
I'm using the following code to read a STEP file:

```typescript
oc.FS.createDataFile('/', fileName, fileText, true, true, true)

const reader = new oc.STEPControl_Reader_1()
const readSuccess = reader.ReadFile(fileName) === oc.IFSelect_ReturnStatus.IFSelect_RetDone

if (!readSuccess) {
  throw 'Failed to convert STEP file to OCCT shape'
}

reader.TransferRoots(new oc.Message_ProgressRange_1())
const shape = reader.OneShape()
oc.FS.unlink(fileName)

const faceExplorer = new oc.TopExp_Explorer_1()

for (
  // @ts-ignore
  faceExplorer.Init(shape, oc.TopAbs_ShapeEnum.TopAbs_FACE, oc.TopAbs_ShapeEnum.TopAbs_SHAPE);
  faceExplorer.More();
  faceExplorer.Next()
) {
  const currentShape = faceExplorer.Current()
  const face = oc.TopoDS.Face_1(currentShape)
  // How to get the STEP instance ID of `face`?
}
```

And let's suppose this data is inside my STEP file:
```
#417=ADVANCED_FACE('face_1',(#112),#405,.F.);
#418=ADVANCED_FACE('face_2',(#113),#406,.F.);
#419=ADVANCED_FACE('face_3',(#114),#407,.F.);
```

How can I read the ID's of the `ADVANCED_FACE` instances? (`#417`, `#418` and `#419`)

---

Maybe it is possible with a `XSControl_TransferReader` (`reader.WS().get().TransferReader().get()`), but I couldn't make it work.

flolu, 2022-08-30:
Solution: https://github.com/tpaviot/pythonocc-core/issues/1119#issuecomment-1231562360

closed 2022-08-30 by flolu

gabrielgriesser, 2023-02-28:
Hello, how did you do ?

I can't find an Opencascade.js code working for this one :(

flolu, 2023-03-04:
Hey @gabrielgriesser. I'm sorry. I only used this with [pythonocc-core](https://github.com/tpaviot/pythonocc-core) and I have never checked if it is also available in Opencascade.js. I recommend you to create a new issue in this repository. Maybe someone from the creators know how to access the `IdentLabel`

gabrielgriesser, 2023-03-07:
Thank you for your answer @flolu 

I used your code with the opencascade.js library, but the function `DownCast` in line `item = StepRepr_RepresentationItem.DownCast(item)` cannot be found in this library.

I presume that this function has not been imported yet.

Here is my code 
```
let anExplorer = new oc.TopExp_Explorer_2(shape, oc.TopAbs_ShapeEnum.TopAbs_FACE, oc.TopAbs_ShapeEnum.TopAbs_SHAPE);

  for (anExplorer.Init(shape, oc.TopAbs_ShapeEnum.TopAbs_FACE, oc.TopAbs_ShapeEnum.TopAbs_FACE); anExplorer.More(); anExplorer.Next()) {
    let item = transferReader.EntityFromShapeResult(anExplorer.Current(), 4);
    console.log("ITEM : " + model.get().IdentLabel(item));
    // print only 0
  }
```

@donalffons have you some tips on it ?