donalffons/opencascade.js #163: Error initializing custom build

MattFerraro, 2022-08-12:
I followed the instructions [here](https://ocjs.org/docs/app-dev-workflow/custom-builds) for building my own custom build, but I used the build definition given [here](https://github.com/donalffons/opencascade.js/blob/master/website/ocjs-editor-theme/src/customBuild/customBuild.yml).

The build went fine, but then when I go to use it:

```
import initOpenCascade from "opencascade.js"
import opencascade from './customBuild/customBuild.examples.js';
import opencascadeWasm from './customBuild/customBuild.examples.wasm';

initOpenCascade({
      mainJs: opencascade,
      mainWasm: opencascadeWasm,
    }).then((newOC) => {
      console.log("wasm is initialized!")
    })
```

The console shows me an error:
```
Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'buffer')
    at receiveInstance (opencascade.full.js:9:1)
    at receiveInstantiationResult (opencascade.full.js:9:1)
```

Expected behavior is that opencascade initializes properly. It *does* init properly if I omit my custom `mainJs` and `mainWasm`, it just takes 5 seconds or so, hence my need for a custom build.

martinRenou, 2023-11-14:
@MattFerraro I can reproduce this error when using `- -sDISABLE_EXCEPTION_CATCHING=0` in the custom build emccFlags, when setting this to `-sDISABLE_EXCEPTION_CATCHING=1` it does work though. I'm unsure what is going on though.

ketanvariya, 2024-02-01:
Hello @MattFerraro,
Have you been able to resolve this problem?

The same error is appearing for me on the same line.

RajanHexa, 2024-02-01:
@ketanvariya No

RajanHexa, 2024-02-01:
@ketanvariya yes it is true to be true