donalffons/opencascade.js #196: Build emcc crashes

kerimcharfi, 2023-01-01:
I can't build it using docker 

```
docker run \
            -v $(pwd):/src \
            -u $(id -u):$(id -g) \
            donalffons/opencascade.js \
            /src/builds/opencascade.full.yml
```

Logs:

```
2023-01-01 16:21:20   /emsdk/upstream/emscripten/system/lib/libcxx/include/__support/musl/xlocale.h:42:34: error: static declaration of 'wcstoull_l' follows non-static declaration
2023-01-01 16:21:20   /emsdk/upstream/emscripten/system/lib/libcxx/include/__support/musl/xlocale.h:48:27: error: static declaration of 'wcstold_l' follows non-static declaration
2023-01-01 16:21:20   /emsdk/upstream/emscripten/system/lib/libcxx/include/locale:752:26: error: call to 'strtoll_l' is ambiguous
2023-01-01 16:21:20   /emsdk/upstream/emscripten/system/lib/libcxx/include/locale:792:35: error: call to 'strtoull_l' is ambiguous
2023-01-01 16:21:34 Diagnostic Messages:
2023-01-01 16:21:34   /emsdk/upstream/emscripten/system/lib/libcxx/include/__support/musl/xlocale.h:27:25: error: static declaration of 'strtoll_l' follows non-static declaration
2023-01-01 16:21:34   /emsdk/upstream/emscripten/system/lib/libcxx/include/__support/musl/xlocale.h:32:34: error: static declaration of 'strtoull_l' follows non-static declaration
2023-01-01 16:21:34   /emsdk/upstream/emscripten/system/lib/libcxx/include/__support/musl/xlocale.h:37:25: error: static declaration of 'wcstoll_l' follows non-static declaration
2023-01-01 16:21:34   /emsdk/upstream/emscripten/system/lib/libcxx/include/__support/musl/xlocale.h:42:34: error: static declaration of 'wcstoull_l' follows non-static declaration
2023-01-01 16:21:34   /emsdk/upstream/emscripten/system/lib/libcxx/include/__support/musl/xlocale.h:48:27: error: static declaration of 'wcstold_l' follows non-static declaration
2023-01-01 16:21:34   /emsdk/upstream/emscripten/system/lib/libcxx/include/locale:752:26: error: call to 'strtoll_l' is ambiguous
2023-01-01 16:21:34   /emsdk/upstream/emscripten/system/lib/libcxx/include/locale:792:35: error: call to 'strtoull_l' is ambiguous
2023-01-01 16:21:35 building /opencascade.js/build/bindings/myMain.h/Handle_IMeshTools_Context.cpp
2023-01-01 16:21:35 building /opencascade.js/build/bindings/myMain.h/OCJS.cpp
2023-01-01 16:21:50 building /opencascade.js/build/additionalBindCode/opencascade.full.js.cpp
2023-01-01 16:21:50 Running build: opencascade.full.js
2023-01-01 16:21:54 ports:INFO: retrieving port: freetype from https://github.com/emscripten-ports/FreeType/archive/version_1.zip
2023-01-01 16:21:55 ports:INFO: unpacking port: freetype

.....

cache:INFO: generating system library: sysroot/lib/wasm32-emscripten/lto/libcompiler_rt.a... (this will be cached in "/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/lto/libcompiler_rt.a" for subsequent builds)
cache:INFO:  - ok
cache:INFO: generating system library: sysroot/lib/wasm32-emscripten/lto/libc++.a... (this will be cached in "/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/lto/libc++.a" for subsequent builds)
cache:INFO:  - ok
cache:INFO: generating system library: sysroot/lib/wasm32-emscripten/lto/libc++abi.a... (this will be cached in "/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/lto/libc++abi.a" for subsequent builds)
cache:INFO:  - ok
cache:INFO: generating system library: sysroot/lib/wasm32-emscripten/lto/libsockets.a... (this will be cached in "/emsdk/upstream/emscripten/cache/sysroot/lib/wasm32-emscripten/lto/libsockets.a" for subsequent builds)
cache:INFO:  - ok
 emcc: error: '/emsdk/upstream/bin/wasm-ld @/tmp/emscripten_f_zc1vqr.rsp.utf-8' failed (received SIGKILL (-9))
Traceback (most recent call last):
 File "/opencascade.js/src/buildFromYaml.py", line 127, in <module>
 runBuild(buildConfig["mainBuild"])
File "/opencascade.js/src/buildFromYaml.py", line 118, in runBuild
subprocess.check_call([
File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
raise CalledProcessError(retcode, cmd)
```

I tried on WSL and Ubuntu github actions

https://github.com/kerimcharfi/opencascade.js/actions/runs/3817610186

frank-pian, 2023-02-06:
I still fail to build with this [process](https://ocjs.org/docs/app-dev-workflow/custom-builds).
```
...
/XCAFDoc_DimTolTool.cxx.o', '/opencascade.js/build/sources/XCAFDoc/XCAFDoc_NotesTool.cxx.o', '/opencascade.js/build/sources/XCAFDoc/XCAFDoc_LayerTool.cxx.o', '/opencascade.js/build/sources/XCAFDoc/XCAFDoc_ShapeTool.cxx.o', '/opencascade.js/build/sources/XCAFDoc/XCAFDoc_VisMaterial.cxx.o', '/opencascade.js/build/sources/GeomAdaptor/GeomAdaptor.cxx.o', '/opencascade.js/build/sources/GeomAdaptor/GeomAdaptor_SurfaceOfLinearExtrusion.cxx.o', '/opencascade.js/build/sources/GeomAdaptor/GeomAdaptor_Surface.cxx.o', '/opencascade.js/build/sources/GeomAdaptor/GeomAdaptor_SurfaceOfRevolution.cxx.o', '/opencascade.js/build/sources/GeomAdaptor/GeomAdaptor_Curve.cxx.o', '/opencascade.js/build/sources/IntStart/IntStart_SITopolTool.cxx.o', '-o', '/src/rocketExample.js', '', '-O3', '-sEXPORT_ES6=1', '-sUSE_ES6_IMPORT_META=0', "-sEXPORTED_RUNTIME_METHODS=['FS']", '-sINITIAL_MEMORY=100MB', '-sMAXIMUM_MEMORY=4GB', '-sALLOW_MEMORY_GROWTH=1', '-sUSE_FREETYPE=1']' returned non-zero exit status 1.
```

donalffons, 2023-02-06:
Hi there, apologies for the late reply. Maybe your system is running out of memory during the build? Your OS would then most likely kill the container. Building the library requires lots of memory - [in the order of ~40-50 GB](https://github.com/donalffons/opencascade.js/issues/103#issuecomment-1084838435). High memory usage seems to be a common issue with highly optimized Emscripten builds.
If you're on Linux, you should be able to configure swap space on your disk to be used when memory runs low. On MacOS / Windows, there should be similar workarounds...

frank-pian, 2023-02-17:
I compiled successfully on a PC with 64GB memory.

kerimcharfi, 2023-02-17:
64 GB and clean Ubuntu also worked for me

closed 2023-02-17 by kerimcharfi