donalffons/opencascade.js #206: MultiThreaded build on the browser gives errors related to node environment

bitbybit-dev, 2023-01-09:
Hi. I followed the steps to create a multi-threaded build and it succeeded, but when trying to start up the app I'm getting two errors

Error: Can't resolve 'perf_hooks'
Error: Can't resolve 'worker_threads'

These two seem to be node environment dependencies, but do not exist in the browser. I wonder if there's something wrong I'm doing when preparing yml file for the build that adds these dependencies. From the use of navigator I suppose that the build should work on the browser somehow.
Or maybe multi-threaded build is not meant to run in the browser? I see that your [example test](https://github.com/donalffons/opencascade.js/blob/master/test/multi-threaded.test.ts) runs in the node directly, where these require statements resolve fine I guess. 
Could you explain this a little? Thanks!

donalffons, 2023-01-17:
Hi, the multi-threaded build does work in the browser. This problematic code comes from Emscripten - check [this SO](https://stackoverflow.com/a/59488387) ticket for two suggestions on how to fix this. If I remember correctly, this issue is a bundler-specific issue (probably webpack?!), where it statically analyzes the code for `require` statements in a build step and then doesn't find those dependencies, when building for the browser. During actual execution however, this code would never be run (Emscripten checks which environment is being used during execution) and would therefore not cause issues.

bitbybit-dev, 2023-01-17:
Ah, thanks for that tip! Will find ways to silence this :)  Hope you're doing well. Was using occ.js quite a bit lately, super powerful.

closed 2023-01-17 by bitbybit-dev