donalffons/opencascade.js #213: Is there a better way to add cpp code?

frank-pian, 2023-02-17:
Adding cpp code to the yml file looks fine.But it's inconvenient when there is a lot of code.
Can I specify that a list of cpp files be included for compilation?

donalffons, 2023-03-05:
I've been thinking of that problem for a while and what I've come up with is implemented in another project called libclangjs, [here](https://github.com/donalffons/libclangjs#extending--integrating-with-other-cc-code). Libclangjs adds JavaScript / WASM bindings to llvm's libclang (similar to how OCJS adds JavaScript / WASM bindings to OCCT - but much simpler). There is a fairly minimal [test](https://github.com/donalffons/libclangjs/tree/main/tests/test-cmake) and some [docs](https://www.npmjs.com/package/libclangjs-cmake) that explain how it works.  In summary:
- The bindings and relevant headers are published as a CMake package, that can be added via CMake's [`find_package`](https://github.com/donalffons/libclangjs/blob/5af1087a3dd29f05b03ca38360d7a07d17c73277/tests/test-cmake/CMakeLists.txt#L7) call to your custom CMake project (which builds an "executable", i.e. the `.js` and `.wasm` files, you want to use in your JS project).
- This CMake package is published via NPM. That might be a bit unusual, but I think it's an easy way of publishing that package, which should integrate nicely with commonly used JS development tools. This should also work nicely with a monorepo layout, where the "custom build" of OCJS is just another package that can be installed by other packages inside that repo.
- C/C++ code can then be written "like usual" in any other CMake-based project.

I believe this is much cleaner and lightweight than the current solution of adding c++ code to a yaml file and starting a huge Docker container for each compilation. If you have any thoughts / opinions on this, please let me know.

PS: This ignores the fact that the custom build yaml-files also specify which bindings should be included. This would be a separate issue under this proposal.

frank-pian, 2023-03-14:
> I've been thinking about this problem for a while and what I've come up with is implemented in another project called libclangjs, [here](https://github.com/donalffons/libclangjs#extending--integrating-with-other-cc-code). Libclangjs adds JavaScript / WASM bindings to llvm's libclang (similar to how OCJS adds JavaScript / WASM bindings to OCCT - but much simpler). There is a fairly minimal [test](https://github.com/donalffons/libclangjs/tree/main/tests/test-cmake) and some [docs](https://www.npmjs.com/package/libclangjs-cmake) that explain how it works. In summary:
> 
>     * The bindings and relevant headers are published as a CMake package, that can be added via CMake's [`find_package`](https://github.com/donalffons/libclangjs/blob/5af1087a3dd29f05b03ca38360d7a07d17c73277/tests/test-cmake/CMakeLists.txt#L7) call to your custom CMake project (which builds an "executable", i.e. the `.js` and `.wasm` files, you want to use in your JS project).
> 
>     * This CMake package is published via NPM. That might be a bit unusual, but I think it's an easy way of publishing that package, which should integrate nicely with commonly used JS development tools. This should also work nicely with a monorepo layout, where the "custom build" of OCJS is just another package that can be installed by other packages inside that repo.
> 
>     * C/C++ code can then be written "like usual" in any other CMake-based project.
> 
> 
> I believe this is much cleaner and lightweight than the current solution of adding c++ code to a yaml file and starting a huge Docker container for each compilation. If you have any thoughts / opinions on this, please let me know.
> 
> PS: This ignores the fact that the custom build yaml-files also specify which bindings should be included. This would be a separate issue under this proposal.

It looks great ! If there are any examples, I would be happy to test.