donalffons/opencascade.js #220: null function or function signature mismatch when building shell from solid

rianadon, 2023-03-04:
This project is some very incredible work! I'm impressed you've brought a whole cad platform to JavaScript.

I'm trying to build a solid object from a shell and have been encountering this error:
```
RuntimeError: null function or function signature mismatch
    at wasm://wasm/0bfe60ea:wasm-function[2629]:0x11fa1d
    at invoke_viii (file://.../node_modules/opencascade.js/dist/opencascade.full.js:9:261824)
    at wasm://wasm/0bfe60ea:wasm-function[199264]:0x23967c3
    at wasm://wasm/0bfe60ea:wasm-function[28983]:0x819c9a
    at ShapeFix_Solid_1.ShapeFix_Solid$SolidFromShell [as SolidFromShell] (eval at new_ (file://.../node_modules/opencascade.js/dist/opencascade.full.js:9:94337), <anonymous>:10:10)
    at hull_faces (.../target/index.cjs:3531:118)
```

The code I'm using is (from Node.js)
```js
const sewing = new oc.BRepBuilderAPI_Sewing(1.0e-06, true, true, true, false)
for (const p of polys) {
  sewing.Add(p)
}
sewing.Perform(new oc.Message_ProgressRange_1())
(new oc.ShapeFix_Solid_1()).SolidFromShell(new oc.TopoDS.Shell_1(sewing.SewedShape())))
```

I'm just starting to learn how to debug web assembly. But based on the stack trace, it looks like there might be some function used by `SolidFromShell` that's not working?

I'm using the latest beta version of opencascade.js on npm.

donalffons, 2023-03-06:
Did you check if the `TopoDS_Shape` returned by `sewing.SewedShape()` is actually a Shell? You could do that by calling the `ShapeType()` method and console.logging / inspecting the return value.

But yeah, this does sound like something is missing in the build... I will look into this at some point, but its probably going to take me a while to get to this.

rianadon, 2023-03-14:
Thank you! I think there's definitely something wrong on my end I did. I used another library wrapping OpenCascade.js to build the shape and it worked without errors.

I'll have to look more into specifically what was wrong about the process of turning the shell into the shape, but at least the library isn't the problem. You're probably right about sewing.SewedShape() returning something unexpected.

That said, your time is probably better spent elsewhere. Thank you for the help!

closed 2023-03-14 by rianadon