donalffons/opencascade.js #222: How to access data from .write()

stefanFuess, 2023-03-13:
I'm trying to access my data from my exported step file.

`  var writer = new openCascade.STEPControl_Writer_1();
  var progressRange = new openCascade.Message_ProgressRange_1();
  writer.Transfer(reader.OneShape(), openCascade.STEPControl_StepModelType.STEPControl_AsIs, true, progressRange);
  writer.Write("test.step")`

no errors are thrown, but I dont know how to access the data now.

How to write my step file?

FANZHUJIONG, 2023-05-30:
We also have problem like this. 
We use opencascade.js to write BREP. Now the question is : though BRepTools.Write_3 return "true", we don't know whether the file "a.brep" is created or not, since we didn't find "a.brep" in the specified file path. The code is below:
```
var shape = createSquare(this.openCascade, 10, 10, 10, 360, 1);
var brepStr = "a.brep";
var theProgress = new this.openCascade.Message_ProgressRange_1();
var result = this.openCascade.BRepTools.Write_3(shape,brepStr,theProgress)
console.log(result)
```
We don't know how to write brep file.

FANZHUJIONG, 2023-05-31:
We found solution. Maybe js doesn't offer methods to write file to disk, and the file that occ.js created is storaged in occ.js buffer or something else, so we can't find file in our disk. But we can read the file and get the String like below:
```
var theProgress = new this.openCascade.Message_ProgressRange_1();
var result = this.openCascade.BRepTools.Write_3(shape,"test",theProgress)
let brep = this.openCascade.FS.readFile("/" + "test", { encoding:"utf8" });
this.openCascade.FS.unlink("/" + "test");
console.log(result)
console.log(brep)
```