donalffons/opencascade.js #268: Cannot bind the library with Vite

maelp, 2023-12-15:
I'm trying to build the library and import it but I keep having errors such as `WebAssembly.instantiate(): Import #0 module="a": module is not an object or function`, I'm not sure why

could you perhaps provide a basic project with a `vite.config.mts` which builds correctly?

mathiasplans, 2023-12-22:
Have a similar problem:
```
 vite-plugin-wasm-namespace:/mnt/c/Users/mathi/Documents/solidgram/node_modules/opencascade.js/dist/opencascade.full.wasm:43:35151:
      43 │ ..._0_1096, ut as __vite__wasmImport_0_1097, vt as __vite__wasmImport_0_1098, wt as __vite__wasmImport_0_1099 } from "a";
         ╵                                                                                                                      ~~~

  You can mark the path "a" as external to exclude it from the bundle, which will remove this error
  and leave the unresolved path in the bundle.
```

ietwangwei, 2024-07-02:
I found a solution with three steps:

1. Use the vite-plugin-wasm plugin, add it into your vit.config

2. Add the wasm file to optimizeDeps.exinclude into vite.config.js
    ```
    exclude: ["opencascade.js/dist/opencascade.full.wasm"]
    ```
3. Add the suffix `?url` to the wasm import path like `opencascade.js/dist/opencascade.full.wasm?url`

wuzejinCN, 2024-10-18:
// import occ with '?url'. Check follow:

import openCascade from 'opencascade.js/dist/opencascade.full.js'
import openCascadeWasm from 'opencascade.js/dist/opencascade.full.wasm?url'

export async function initOcc() {
  return await openCascade({
    locateFile: () => openCascadeWasm,
  })
}

tetron432, 2025-06-17:
I added this in my vite.config.js: 
```
optimizeDeps: {
     exclude: ["opencascade.js/dist/opencascade.full.wasm"],
},
```

and did:

`import initOpenCascade from "opencascade.js/dist/opencascade.full.wasm?url"`

and I still get errors like:

`SES_UNCAUGHT_EXCEPTION: TypeError: "/node_modules/opencascade.js/dist/opencascade.full.wasm" is not a function`

Can someone help?

tetron432, 2025-06-20:
> [@tetron432](https://github.com/tetron432) Have you tried [#268 (comment)](https://github.com/donalffons/opencascade.js/issues/268#issuecomment-2422570040)? That should work

I have tried that, and while it didn't throw any errors, the library becomes incredibly sluggish.

Ruud-DRS, 2025-06-26:
> > [@tetron432](https://github.com/tetron432) Have you tried [#268 (comment)](https://github.com/donalffons/opencascade.js/issues/268#issuecomment-2422570040)? That should work
> 
> I have tried that, and while it didn't throw any errors, the library becomes incredibly sluggish.

Could you specify sluggish?