donalffons/opencascade.js #270: Update opencascade to 7.8.0

trungleduc, 2024-01-08:
Hi @donalffons, thank you very much for this awesome project!
Currently `opencascade.js` pinned `opencascade`  to 7.6.2. Do you happen to have a plan to update it to a more recent version? 
If yes, I'm ready to work on it if you can give me some pointers to start.

ninja-, 2024-01-09:
👁️

Milesssssss, 2024-06-04:
👁️

jwatzman, 2025-08-13:
Updating to version 7.6.3 is extremely straightforward -- I didn't open a PR for it for whatever reason, but https://github.com/jwatzman/opencascade.js/commit/e1da85a792d5a7cb4fa471b7ce48449288b5bba2 will do it, and I think that will apply cleanly to this upstream repo. (Maybe I didn't open a PR because it also does a cleanup around environment variables and `tar`?)

I'm not working with opencascade.js any more, but when I was (and playing with my fork above), I think I tried to update to 7.8.x and it was a more complicated update. I don't remember why. I think trying to get that to work is what motivated #292, so that I could actually iterate on changes in a reasonable timeframe, but then the prospective project this was for ended up deciding not to use opencascade.js and so I stopped there.

cesarecaoduro, 2025-08-13:
I absolutely gave up on the topic of upgrading opencascadejs, tried with all I know and Claude Code knows, and we both failed :)