donalffons/opencascade.js #273: Project Maintenance Status?

jteppinette, 2024-03-05:
Does anyone know the maintenance status of this project. Is it time to investigate a fork since there hasn't been a published release in a year, and it is beginning to lag significantly behind OCCT.

goswinr, 2024-03-06:
I wonder too,
these two forks are ahead of master:
https://github.com/TheTechCompany/opencascade.js
https://github.com/ashmna/opencascade.js

clinuxrulz, 2024-06-27:
Another way forward is to design a web worker in C++ that your web app talks to using emscripten and OpenCascade (C++ version) and only embind/expose your message processing function for your web worker instead of exposing the entire opencascade API. The message processing function is just a function that takes a string and returns a string.

The final wasm build from emscripten would then only contain the wasm code for the opencascade methods (and their dependencies) you touched in your message processing functions for your web worker. (Smaller wasm file.)

clinuxrulz, 2024-06-27:
Here a sample for reference:

https://dev.opencascade.org/doc/overview/html/occt_samples_webgl.html

You'd just need to be happy with writing your Web Worker in C++ is all.

clinuxrulz, 2024-07-02:
This is what I mean:

https://github.com/clinuxrulz/psf-toolbox-occt-webworker/blob/main/main.cpp

Rather than make your web worker in JavaScript/Typescript and embind/ffi a hundred thousand functions.
Just write your webworker in C++ using the latest official opencascade, then compile it all to wasm using emscripten and only embind/ffi the one function that takes and returns a string (```std::string process_message(std::string message);```)

The file size went from a 50mb wasm down to a 30mb wasm. The wasm file will auto-shrink to only include what functions ```process_message``` depends on, and you still get full access to everything in opencascade while coding.

cesarecaoduro, 2024-09-12:
@clinuxrulz this looks interesting...
Any progress on the topic?

clinuxrulz, 2024-09-14:
In honesty I went back to just using opencascade.js , because the debugging your code that uses opencascade is way easier. openscade.js suits my needs for now.

But the "write your web worker in C++" does work. I tested it, and it works well.